gazebosim/ros_gz #26: Trouble launching GPU_lidar

BriceRenaudeau, 2019-07-10:
Hello,

The demo gpu_lidar.launch doesn't work.
The diff_drive and air pressure demos does work.
[image]

I tried using "ogre" instead of "ogre2" for the rendering (it worked for other demos like depth cameras) but the trouble persists :   

> ruby: /build/ogre-1.9-B6QkmW/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreSingleton.h:80: Ogre::Singleton<T>::Singleton() [with T = Ogre::LogManager]: Assertion `!msSingleton' failed.

The full console print is attached [gpu_lidar_console.txt](https://github.com/osrf/ros1_ign/files/3376561/gpu_lidar_console.txt)

I am trying to transfer my simulation from Gazebo to Ignition gazebo.
I installed Igntion Gazebo and this package following your documentation.
I am using ROS Melodic with Ubuntu 18.04.1

chapulina, 2019-07-10:
I've seen the `Unable to create the rendering window` error before. That usually means you're using an old version of OpenGL or have an old Nvidia driver. Could you check:

* `~/.ignition/rendering/ogre2.log` and see if you have `Failed to create an OpenGL 3+ context`
* Your OpenGL version, it should be 3+: `glxinfo | grep "OpenGL version"`
* Your Nvidia driver version, it should be 384+: `nvidia-smi`

BriceRenaudeau, 2019-07-10:
Thanks for your answer,
* In the log, i have this : 

> 14:29:01: Installing plugin: GL 3+ RenderSystem
> 14:29:01: OpenGL 3+ Rendering Subsystem created.
> 14:29:02: Plugin successfully installed

* OpenGL version is : `OpenGL version string: 3.0 Mesa 18.2.8`

* I installed  `nvidia-384`, `nvidia-utils-390` and `nvidia-driver-390` but `nvidia-smi` couldn't communicate with the NVIDIA driver.

Now in the .ignition log I got : 

> OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers.

But I did update it. My ubuntu kernel is 4.15.0-54. Is it possible that I need to upgrade ubuntu to the latest kernel?

chapulina, 2019-07-10:
If `nvidia-smi` didn't work, I think the problem is that you don't have the correct drivers enabled. Try switching drivers with the "Aditional Drivers" dialog?

BriceRenaudeau, 2019-07-10:
`dpkg -l | grep nvidia` is giving me : 

> ii  libnvidia-cfg1-390:amd64
> ii  libnvidia-common-390 
> ii  libnvidia-compute-390:amd64  
> ii  libnvidia-compute-390:i386       
> ii  libnvidia-decode-390:amd64    
> ii  libnvidia-decode-390:i386       
> ii  libnvidia-encode-390:amd64    
> ii  libnvidia-encode-390:i386       
> ii  libnvidia-fbc1-390:amd64       
> ii  libnvidia-fbc1-390:i386   
> ii  libnvidia-gl-390:amd64     
> ii  libnvidia-gl-390:i386  
> ii  libnvidia-ifr1-390:amd64     
> ii  libnvidia-ifr1-390:i386           
> rc  nvidia-340             
> ii  nvidia-384                  
> ii  nvidia-compute-utils-390    
> ii  nvidia-dkms-390       
> ii  nvidia-driver-390     
> ii  nvidia-kernel-common-390       
> ii  nvidia-kernel-source-390 
> rc  nvidia-opencl-icd-340    
> ii  nvidia-prime        
> ii  nvidia-settings        
> ii  nvidia-utils-390  
> ii  xserver-xorg-video-nvidia-390   

`sudo lshw -C display` gives me  : 

>  *-display                 
>        description: VGA compatible controller
>        product: Intel Corporation
>        vendor: Intel Corporation
>        physical id: 2
>        bus info: pci@0000:00:02.0
>        version: 00
>        width: 64 bits
>        clock: 33MHz
>        capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
>        configuration: driver=i915 latency=0

I don't know how to set nvidia driver.

chapulina, 2019-07-10:
> There is nothing in the "additonal Drivers" dialog.

Setting up drivers can be complicated, I hope you figure it out.

I think we've established that the issue is in your local setup, and not a bug in this package, so I'm going to close this issue. Feel free to ask for help at http://answers.gazebosim.org/. Good luck :crossed_fingers:

closed 2019-07-10 by chapulina