gazebosim/ros_gz #27: URDF Spawner

BriceRenaudeau, 2019-07-10:
Hello,
I am trying to use Gazebo ignition instead of gazebo and I would like to know if there exists a model spawner that takes urdf input like in gaezbo-ros ?

chapulina, 2019-07-10:
You can spawn URDF just like you would spawn SDF using just Ignition, you can try:

`ign gazebo empty.sdf`

then

```
ign service -s /world/empty/create \
--reqtype ignition.msgs.EntityFactory \
--reptype ignition.msgs.Boolean \
--timeout 300 \
--req 'sdf: '\
'"<?xml version=\"1.0\" ?>'\
'<robot name=\"will_be_ignored\">'\
'<link name=\"link\">'\
'<visual>'\
'<geometry>'\
'<sphere radius=\"1.0\"/>'\
'</geometry>'\
'</visual>'\
'<collision>'\
'<geometry>'\
'<sphere radius=\"1.0\"/>'\
'</geometry>'\
'</collision>'\
'<inertial>'\
'<mass value=\"1\"/>'\
'<inertia ixx=\"1\" ixy=\"0.0\" ixz=\"0.0\" iyy=\"1\" iyz=\"0.0\" izz=\"1\"/>'\
'</inertial>'\
'</link>'\
'</robot>" '\
'pose: {position: {z: 10}} '\
'name: "new_name" '\
'allow_renaming: true'
```

This will work for the simulation side. If you need the URDF for RViz for example, you'll need to set the `robot_description` parameter separately.

There isn't a pure ROS way of calling the Ignition service yet

BriceRenaudeau, 2019-07-10:
I can spawn my robot using _robot.sdf_ but I would like to know if I can use my previous _robot.gazebo_ file and its plugins.

chapulina, 2019-07-10:
What format does the `.gazebo` file use? If it's SDF or URDF then it should work.

Plugins will not work though. Ignition's C++ API is completely different from classic Gazebo, so all plugins must be rewritten.

BriceRenaudeau, 2019-07-10:
It's SDF format using <robot> balise.
What is the command line to spawn from a file ?

chapulina, 2019-07-10:
Use the `sdf_filename` field. The whole message is documented [here](https://bitbucket.org/ignitionrobotics/ign-msgs/src/default/proto/ignition/msgs/entity_factory.proto).

BriceRenaudeau, 2019-07-10:
> What's "balise"?

Sorry I meant `<robot >` Markup

> Use the sdf_filename field.
 
I tried but it gives me an error :

> [libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format ignition.msgs.EntityFactory: 1:5: Message type "ignition.msgs.EntityFactory" has no field named "from".

Thanks

chapulina, 2019-07-10:
> Sorry I meant `<robot >` Markup

The `<robot>` tag is present on URDF but not on SDF. I suggest you go through some [tutorials](http://gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros).

> I tried but it gives me an error

It's hard to tell what's going on without knowing the exact command you sent. How about we follow up on Gazebo Answers? You can ask a new question detailing what you're trying to do and what steps you've taken.

I'll leave this issue here since it's a valid request for a ROS interface to the entity create service.

BriceRenaudeau, 2019-07-10:
Sorry for bothering you.
I succeeded in using a file.
I still have trouble going from my previous Gazebo file to the new Ignition one.
For exemple : 
* `<color rgba="1.0 1.0 1.0 0.5"/>` syntax doesn't work
* inertia can be wrong now

Thanks.

chapulina, 2019-07-10:
http://answers.gazebosim.org/ is the best place to follow up on these issues :wink:

caguero, 2019-08-02:
I'm closing this one, feel free to follow-up on http://answers.gazebosim.org/ .

closed 2019-08-02 by caguero

chapulina, 2019-11-21:
#51 adds an executable to spawn SDF / URDF which can be used inside launch files and take the model description from a ROS param.