gazebosim/ros_gz #31: parameter_bridge echos back when it receives ROS messages

azeey, 2019-08-01:
When publishing to the ROS side of the parameter_bridge node, it echos back the message. To test, with roscore running, in one terminal I ran
```
rosrun ros1_ign_bridge parameter_bridge test@std_msgs/String@ignition.msgs.StringMsg
```
In another (terminal 2),
```
rostopic echo /test
```

And finally
```
rostopic pub /test std_msgs/String "data: 'Hello'"
```

I expect to see one message in terminal 2, but I get:

```
data: "Hello"
---
data: "Hello"
---
```

caguero, 2019-08-01:
Shouldn't you add `--once` when publishing the message?

azeey, 2019-08-01:
I tried with `--once` and got the same result.

chapulina, 2019-08-01:
I can reproduce it. And ignition only gets the message once:

```
$ ign topic -e -t /test
data: "Hello"
```

caguero, 2019-08-01:
Me too and as @chapulina mentioned it should only happen in the ROS->Ign direction. I think I know what's happening. I might need to touch Ignition Transport to avoid this issue here.

caguero, 2019-08-01:
I have a potential fix in #33 .

caguero, 2019-08-02:
Fixed in pull request #33 .

closed 2019-08-02 by caguero