gazebosim/ros_gz #63: Feature Request: Add Float64 support in ros_ign_bridge

realdealneil, 2020-03-24:
We are trying to implement a pan/tilt mechanism in ignition-gazebo (for SubT).  We have the pan/tilt working using ignition messages, but the ignition joint controller expects an ignition::msgs::Double and ros_ign_bridge does not currently support double conversion from std_msgs::Float64 to ignition::msgs::Double.  

I am starting to go through the ros_ign_bridge code to see how difficult it will be to implement this conversion.  Is there any reason that this hasn't yet been implemented?  An alternative place to fix this would be in ign-gazebo joint controller where we could have it accept a float instead of a double...any reason to prefer one fix over the other?

chapulina, 2020-03-24:
>  Is there any reason that this hasn't yet been implemented?

I think it just hasn't been needed so far.

> any reason to prefer one fix over the other?

Changing the joint controller would not be backwards compatible, so I recommend adding the message conversion here if that solves the issue.

> how difficult it will be to implement this conversion

Adding a new conversion takes a lot of boilerplate code, but it's not too complicated. See https://github.com/osrf/ros_ign/pull/50/ for an example.

realdealneil, 2020-03-24:
https://github.com/osrf/ros_ign/pull/64

chapulina, 2020-03-26:
#64

closed 2020-03-26 by chapulina