gazebosim/ros_gz #76: How to release and run different flavours of ros_ign packages

j-rivero, 2020-05-13:
The original request is from @chapulina. This issue speaks only about the issue of having binary packages supporting different ROS distros and different Ignition distros. A source build for these combinations seems to be possible after changes in pr #72 

A release of `ros_ign` code should ideally fulfill the following requirements:
 1. Package names in code repository should be maintain
   *i.e: do not modify `ros_ign` or `ros_ign_bridge`*
 1. Multiple ROS distributions and multiple ROS packages for the different distributions
   *i.e: `ros-melodic-ros-ign` and `ros-noetic-ros-ign`*
 1. Multiple Ignition versions and multiple packages for the different ignition versions
   *i.e: `ros-melodic-ros-ign-citadel` and `ros-melodic-ros-ign-blueprint`*
 1. Packages must be co-installable
   *problem: filesystem paths are the same within the same ROS distribution since package names are the same*
 1. Packages should install ROS code in the way that versions can be selected at launch/runtime
   *problem: how to rosrun/roslaunch different code if package name is the same*

j-rivero, 2020-05-13:
Without touching how the code is organized in the repo right now:
 * 1, 2 and 3 can be implemented using a bit of bloom templates magic to change debian package names
 * 4 and 5 affects where the code will be in the filesystem

Without asking the user for changing `ROS_PACKAGE_PATH` I don't find a way of having 4 implemented without altering the file paths (include/, lib/ or share/ for example).

The 5 is even more difficult without implementing a wrapper in the source code that accepts an input about which Ignition version the user wants to run and loads different code. Combining both, different filesystem location for one the packages and the wrapper (altering `ROS_PACKAGE_PATH` at runtime to find that different location) we could try to get all the points.

chapulina, 2020-06-17:
tl;dr Let's drop requirements 4 and 5 and have only one `ros_ign` version at a time per ROS distro.

---

Some more considerations:

* Citadel is in https://packages.ros.org for Focal / Buster (https://github.com/ros/rosdistro/pull/24934)
* Each ROS version will support an official Ignition version, the same way as it supports an official Gazebo version:

    * [REP-2000](https://www.ros.org/reps/rep-2000.html#foxy-fitzroy-may-2020-may-2023): Foxy - Citadel
    * [REP-0003](https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025): Noetic - Citadel

* We should release `ros_ign` for the officially supported version on https://packages.ros.org. Right now, this means:

    * `ros-noetic-ros-ign` is Citadel
    * `ros-foxy-ros-ign` is Citadel

* We could also retroactively make it official that ROS Melodic goes with Ignition Blueprint (the current combination used by SubT), so:

     * `ros-melodic-ros-ign` is Blueprint

* To prevent creating multiple sets of `ros_ign` packages that don't talk to each other, we should **not** release unofficial versions upstream. i.e. we shouldn't have any Blueprint / Dome packages in https://packages.ros.org for Focal.

* We should offer alternative non-official combinations for users who really know what they're doing. Those would be released into https://packages.osrfoundation.org like [unofficial](http://gazebosim.org/tutorials?tut=ros_wrapper_versions#UsingaspecificGazeboversionwithROS) `gazebo_ros_pkgs` versions. This could break point 4 above, "_Packages must be co-installable_". I think it wouldn't be too terrible if we said that  `ros-noetic-ros-ign` conflicts with ` ros-noetic-ros-ign-dome`, because we should really be encouraging users to use the official combination.

* If we drop point 4 like suggested above, then point 5 disappears.

---

What do you think, @j-rivero ?

chapulina, 2020-07-22:
The team has discussed this a while back and I believe there are no objections.

Here's the plan:

# ROS 1

- [ ] Support choosing Blueprint / Citadel for Melodic: https://github.com/ignitionrobotics/ros_ign/pull/72
- [ ] Add Noetic support (Citadel only): https://github.com/ignitionrobotics/ros_ign/pull/93
- [ ] Clean up installed headers for Melodic: https://github.com/ignitionrobotics/ros_ign/pull/96
- [ ] Port :point_up: to Noetic
- [ ] Bloom Noetic to https://packages.ros.org with Citadel support
- [ ] Bloom Melodic to https://packages.osrfoundation.org with Blueprint support

# ROS 2

- [ ] Support choosing Blueprint / Citadel for Dashing / Eloquent (port #72)
- [ ] Update Foxy CI to use Citadel from https://packages.ros.org: https://github.com/ignitionrobotics/ros_ign/pull/94
- [ ] Clean up installed headers for Dashing / Eloquent: https://github.com/ignitionrobotics/ros_ign/pull/95
- [ ] Port :point_up: to Foxy
- [ ] Bloom Foxy to https://packages.ros.org with Citadel support

chapulina, 2020-07-23:
> I think we need a place to store this information (official vs unofficial releases, naming, versioning, etc.).

Good point, I'll open a PR documenting this within this repo.

>  the first number gets a meaning

I like that the numbers will have meaning. This means we won't be using semver though. So we should document what minor and patch bumps mean, as well as what's our API/ABI policy.

I have a slight preference for bumping a major for each ROS distro so we can piggyback on [ROS 2's versioning guidelines](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#versioning).

Another thing to consider is not to bump above 1.0 yet so we don't commit to API stability. In that case, we could keep 0.X across all distros, and use different minors to differentiate them, for example:

 * Melodic: 0.10.X
 * Noetic: 0.11.X
 * Dashing: 0.20.X
 * Eloquent: 0.21.X
 * Foxy: 0.22.X

j-rivero, 2020-07-23:
> 
> I like that the numbers will have meaning. This means we won't be using semver though. So we should document what minor and patch bumps mean, as well as what's our API/ABI policy.
>

Oh I assumed that in ROS, specially in ROS1, we don't quite follow semver.
 
> I have a slight preference for bumping a major for each ROS distro so we can piggyback on [ROS 2's versioning guidelines](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#versioning).

Good resource indeed.

> Another thing to consider is not to bump above 1.0 yet so we don't commit to API stability. In that case, we could keep 0.X across all distros, and use different minors to differentiate them, for [...]

Like it, deal.

chapulina, 2020-07-23:
@j-rivero , I documented some of the things discussed here on #99. Let me know what you think. I kept the version numbers at 0.X, but went a bit crazy and added one more minor digit :smile:

j-rivero, 2020-07-24:
>  Bloom Melodic to https://packages.osrfoundation.org with Blueprint support

We currently host packages for ros_ign bridge in Melodic:
```
jrivero@nium ~ $ apt-cache show ros-melodic-ros-ign-bridge 
Package: ros-melodic-ros-ign-bridge
Version: 0.9.3-3bionic
Depends: libc6 (>= 2.14), libgcc1 (>= 1:3.0), libignition-msgs4, libignition-transport7, libprotobuf10, libstdc++6 (>= 5.2), libignition-msgs4-dev, libignition-transport7-dev, ros-melodic-geometry-msgs, ros-melodic-mav-msgs, ros-melodic-message-runtime, ros-melodic-nav-msgs, ros-melodic-rosconsole, ros-melodic-roscpp, ros-melodic-rosgraph-msgs, ros-melodic-sensor-msgs, ros-melodic-std-msgs, ros-melodic-std-srvs, ros-melodic-tf2-msgs
```
For this new release we would be changing names to `ros-melodic-ign-*-blueprint`, right? We want the current packages to be automatically updated to new names?

chapulina, 2020-07-25:
> For this new release we would be changing names to ros-melodic-ign-*-blueprint, right?

Oh no, I think that would break SubT. I think we agreed that `ros-melodic-ros-ign` will be compiled against Blueprint and we won't provide Citadel debs for now.

closed 2020-09-09 by chapulina