gazebosim/ros_gz #109: How to load plugin from URDF in ignition ?

doisyg, 2020-09-08:
Transitioning from Gazebo to Ignition, I am trying to use the spawner https://github.com/ignitionrobotics/ros_ign/pull/60
I can successfully load an URDF which spawn a model in Ignition, but in don't understand how to load a plugin from the URDF.
More precisely I am trying to start the DiffDrive plugin, I put the following :

      <plugin
        filename="libignition-gazebo-diff-drive-system.so"
        name="ignition::gazebo::systems::DiffDrive">
        <left_joint>left_wheel_joint</left_joint>
        <right_joint>right_wheel_joint</right_joint>
        <wheel_separation>1.25</wheel_separation>
        <wheel_radius>0.3</wheel_radius>
        
      </plugin>

below under **robot** tag but nothing seems to be loaded. However when using a sdf file and putting it under the **model** tag, it works properly. Maybe there is an equivalent to the urdf gazebo tag for ignition but I cannot find any example.

diegoferigo, 2020-09-08:
Have you tried the following?

```xml
<robot name="model_name">
  <gazebo>
    <plugin filename="" ...>
      <!-- ... -->
    </plugin>
  </gazebo>
</robot>
```

The urdf file is converted by gazebo to sdf before being parsed by sdformat. You can test the conversion with `ign sdf -p file.urdf`.

doisyg, 2020-09-08:
:man_facepalming: :man_facepalming: :man_facepalming:
Thanks, it works!
and `ign sdf -p file.urdf` is very useful for debugging

closed 2020-09-08 by doisyg

diegoferigo, 2020-09-08:
For future reference, the [URDF tutorial](http://gazebosim.org/tutorials?tut=ros_urdf) for Gazebo Classic seems still valid also for Ignition Gazebo. In particular, from the section [`<gazebo> element for the tag`](http://gazebosim.org/tutorials?tut=ros_urdf#%3Cgazebo%3Eelementforthetag):

> If a `<gazebo>` element is used without a `reference=""` property, it is assumed the `<gazebo>` element is for the whole robot model.

that can be applied also to model plugins.