gazebosim/ros_gz #112: Remap in ros_ign_bridge ?

doisyg, 2020-09-20:
In order to be able to set the output ROS topic of my simulated depth sensor confgurable (see https://github.com/ignitionrobotics/ign-gazebo/issues/360), I thought about having the possibility to have a different topic name on the ign side and on the ROS side.
Would this be useful or create some confusion?

What about using the  `:=` operator with for instance, the following syntax:
`/chatter_in_ign:=/chatter_in_ros@std_msgs/String@ignition.msgs.StringMsg`

chapulina, 2020-09-20:
I don't think this is a bad idea, but I think you should already be able to achieve this using ROS' remapping, right? See [this example](https://github.com/osrf/subt/blob/0eed307c9b25bc0cfdf3a1577b090fa64926c384/submitted_models/explorer_x1_sensor_config_1/launch/vehicle_topics.launch#L9-L15). You can also take advantage of namespaces this way.

Since there's already a way to do it, I'd lean towards not adding yet another remapping mechanism.

doisyg, 2020-09-21:
You are right, using ROS remapping is much simpler!

closed 2020-09-21 by doisyg