gazebosim/ros_gz #114: ros_ign_bridge error for nest model sensor topic

gezp, 2020-10-02:
I create a nest model sdf which contains a gpu_lidar sensor with default topic, then i  publish this topic  to ros by using use ros_ign_bridge.but i get a error like this.so how to solve it?  i have no idea now, thank you.

```
what():  Invalid topic name: topic name must not contain characters other than alphanumerics, '_', '~', '{', or '}':
```
this is topic info i use `ign topic -l` to get
```
/world/demo/model/my_robot/link/front_lidar::rplidar_a2/sensor/lidar_sensor/scan
```

chapulina, 2020-10-02:
Nested models aren't fully supported in Ignition yet, see this PR in the works: https://github.com/ignitionrobotics/ign-gazebo/pull/258

Do you have simple world / launch files that reproduce your issue?

Also, what Ignition / ROS versions are you using?

gezp, 2020-10-03:
I use ros2 foxy and ignition dome. now i know the reason of the error, because the characters ` "::"`  is valid in topic name for `Ignition-transport` ,but is invalid for `ROS` .i solved this by modifying the source code of `ros_ign/ros_ign_bridge`  ros2 branch.

First, I copy `replace_delimiter()` function in `convert.cpp` to `parameter_bridge.cpp`, which is used to replace `::` with `/`  so that make topic name of ros valid. Then, I modify the case` FROM_IGN_TO_ROS` in main(). (i only change parameter_bridge.cpp)

Before:
```
        case FROM_IGN_TO_ROS:
          ign_to_ros_handles.push_back(
              ros_ign_bridge::create_bridge_from_ign_to_ros(
                ign_node, ros_node,
                ign_type_name, topic_name, queue_size,
                ros_type_name, topic_name, queue_size));
          break;
```
After:
```
        case FROM_IGN_TO_ROS:
          ign_to_ros_handles.push_back(
              ros_ign_bridge::create_bridge_from_ign_to_ros(
                ign_node, ros_node,
                ign_type_name, topic_name, queue_size,
                ros_type_name, replace_delimiter(topic_name, "/", "::"), queue_size));
          break;
```

chapulina, 2020-10-12:
@gezp , do you have a minimal example of a world that results in topics with `::` in them? 

I think that ideally we should be substituting the `::` for `/` from topics at the `ign-gazebo` level.

closed 2021-12-07 by gezp