gazebosim/ros_gz #115: lidar can is infinite

Nitrow, 2020-10-10:
Hello, I don't know if this is the please to ask questions, but here it goes.
When I publish a lidar scan over the ignition ros bridge it gives me "inf" if there is no obstacle. I was wondering what the easiest way to just have it publish the max lidar range when this is the case. Pretty important since I want to simulate gmapping. I could create a note that subscribes to the ros topic and change the values and then publish it on another topic but that seems like a lot of work for such a minor thing. 

Cheers.

Nitrow, 2020-10-11:
Never mind, I just did the annoying way.

closed 2020-10-11 by Nitrow

chapulina, 2020-10-12:
>  I don't know if this is the please to ask questions, but here it goes.

https://answers.gazebosim.org/ or https://answers.ros.org/ are better places for questions :wink: 

> a lidar scan over the ignition ros bridge it gives me "inf" if there is no obstacle.

We're following [REP-117](https://www.ros.org/reps/rep-0117.html):

"_Detections that are too close to the sensor to quantify shall be represented by -Inf. Erroneous detections shall be represented by quiet (non-signaling) NaNs. Finally, out of range detections will be represented by +Inf._"