gazebosim/ros_gz #116: TF2 msgs not implemented in ros2 branch

09ubberboy90, 2020-10-15:
PR #67 was merged in the melodic branch but has not been ported to the ROS2 branch.
This means there is no way to send tf messages to ignition for Ros2

Is there any plans to merge the melodic branch in the Ros2 branch sometime soon ?

If not I'm going to try and port in and will do a pull request

chapulina, 2020-10-15:
> Is there any plans to merge the melodic branch in the Ros2 branch sometime soon ?

Unfortunately porting between ROS versions is not that straightforward and requires a separate pull request.

> If not I'm going to try and port in and will do a pull request

That would be great, thanks!

09ubberboy90, 2020-10-15:
> That would be great, thanks!
Cool I'm gonna try and port that feature. Not sure about doing a full merge as I've never done that.

chapulina, 2020-10-15:
> Not sure about doing a full merge as I've never done that.

Yeah the merge wouldn't be a good idea. You can try cherry-picking 7b06c210379680bfa96f3e1d7794708f0e97007e to a ROS 2 branch (I suggest `dashing` if possible, otherwise `ros2`), then manually fixing the differences.

09ubberboy90, 2020-10-22:
I can close this right?

chapulina, 2020-10-23:
Done in https://github.com/ignitionrobotics/ros_ign/pull/117, thanks @09ubberboy90 !

closed 2020-10-23 by chapulina