gazebosim/ros_gz #118: Ros->Ign Bridge stops with signal 245 when JointStatePublisher is run

09ubberboy90, 2020-10-22:
If you run a bridge to publish joint_states from Ros2 to ignition with `ros run ros_ign_bridge parameter_bridge /joint_states@sensor_msgs/msg/JointState@ignition.msgs.Model` it works. 

However if you run JointStatePublisher or JointStatePublisherGui with `ros run joint_state_publisher joint_state_publisher` and a Urdf file the bridge stops with no visible error beside signal 245 beeing sent
[image]

chapulina, 2020-10-26:
Thank you for ticketing the issue. Do you have a minimal launch file that can be used to test this issue?

Also, what Ignition version are you using?

09ubberboy90, 2020-10-31:
Hi sorry for the delay. I've been pretty busy. I am using ignition 4.0.0. I also have ignition 3.3.0 installed.

A simple launch file would be (once given a pkg and an urdf file with joints inside):

``` python
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    pkg_name = "PKG_NAME"
    urdf_file = "URDF_FILE"
    pkg_share = get_package_share_directory(pkg_name)

    joint = Node(
        package='joint_state_publisher_gui',
        executable='joint_state_publisher_gui',
        output='screen',
        arguments=[os.path.join(pkg_share, urdf_file)],
    )

    bridge = Node(
        package='ros_ign_bridge',
        executable='parameter_bridge',
        arguments="/joint_states@sensor_msgs/msg/JointState@ignition.msgs.Model",
        output='screen'
    )

    return LaunchDescription([
        joint,
        bridge,
    ])
``` 
This causes an causes an `[ERROR] [parameter_bridge-2]: process has died [pid 194342, exit code 255, cmd '/home/ubb/Documents/ros/bridge_ws/install/ros_ign_bridge/lib/ros_ign_bridge/parameter_bridge / j o i n t _ s t a t e s @ s e n s o r _ m s g s / m s g / J o i n t S t a t e @ i g n i t i o n . m s g s . M o d e l --ros-args']`

along with the documentation of how to properly format the arguments. This seems to be due that the argument is malformed which is not the case in the launch file.

However running the same command in terminal doesn't seem to generate such an error (Nothing else is running). And the bridge only dies once the JointStatePublisher starts to publish. Not providing an urdf file doesn't crash the bridge as there is nothing beeing publish. So the error is probably in either the subscriber for JointStates or in the publisher for Model

[image]

mcaniot, 2021-12-13:
I have the same problem using ignition citadel and ros foxy. When you launch the ign_bridge alone, there is no error message. You launch a node joint_state_publisher, an error message appear with command parameter_bridge. Same message using the demo code from ignition_bridge/ros_ign_gazebo_demos joint_states.launch.py.
`[ERROR] [parameter_bridge-3]: process has died [pid 1841464, exit code -11, cmd '/home/Documents/test/install/ros_ign_bridge/lib/ros_ign_bridge/parameter_bridge /clock@rosgraph_msgs/msg/Clock@ignition.msgs.Clock /world/empty/model/rrbot/joint_state@sensor_msgs/msg/JointState@ignition.msgs.Model --ros-args -r /world/empty/model/rrbot/joint_state:=joint_states'].`

09ubberboy90, 2021-12-13:
If I remember correctly I think it was due that in ignition the joint topic for control publishes to `/model/{robot_name}/joint/{topic_name}/0/cmd_pos` which causes problem because ros doesn't allow topic to start with a number. But then again I may have encountered this problem when trying to find a workaround.

Anyway until it is fixed I would recommend you create some sort of republisher that listen to joint_states in ros and publishes it in ignition. [Here](https://github.com/09ubberboy90/ros2_sim_compare/blob/c76c696e987e59d7e4fd1c2f1c17c6353492cb3a/simple_arm_controller/src/republisher.cpp) is my implementation of it. With some tweaking you can get something up and running quite fast. 

Hope it helps