gazebosim/ros_gz #119: Failed to load xml from ROS param [ros_ign_gazebo create]

chama1176, 2020-10-23:
## Test environment
- Ubuntu 20.04
- ROS2 Foxy
- ignition citadel

## Test launch file
```
<?xml version="1.0"?>
<launch>
  <group>
    <include file="$(find-pkg-share ros_ign_gazebo)/launch/ign_gazebo.launch.py">
      <arg name="ign_args" value="empty.sdf" />
    </include>
  </group>

  <node pkg="ros_ign_gazebo" exec="create" args="-world empty -param robot_description" output="screen">
    <param name="robot_description" value="$(command 'cat $(find-pkg-share test_robot_ign_gazebo)/urdf/my_robot.urdf')" />
  </node>
</launch>
```
This failed for "Failed to get XML from param".

## How I resolved
### Before
https://github.com/ignitionrobotics/ros_ign/blob/94a719adcf6005fae657a174489cb99349c33ced/ros_ign_gazebo/src/create.cpp#L44-L52

### After
```
   ros::init(_argc, _argv, "ign_create"); 
   ros::NodeHandle nh; 

   gflags::AllowCommandLineReparsing(); 
   gflags::SetUsageMessage( 
       R"(Usage: create -world [arg] [-file FILE] [-param PARAM] [-string STRING] 
                        [-name NAME] [-X X] [-Y Y] [-Z Z] [-Roll ROLL] 
                        [-Pitch PITCH] [-Yaw YAW])"); 
   gflags::ParseCommandLineFlags(&_argc, &_argv, true);   
```
This change works good for me.

chapulina, 2020-10-26:
Thank you for ticketing the issue, @chama1176 . I haven't tried reproducing the issue, but your proposed fix looks ok to me. Would you be able to open a pull request with the fix?

chama1176, 2020-10-27:
Thank you for confirmation.
OK.I will send a pull request with the fix.

chapulina, 2020-10-29:
Should be fixed by #120

PR #120 (2020-10-29): Changed for loading xml from ROS param(#119)
I have fixed #119
closed 2020-10-29 by chapulina