gazebosim/ros_gz #127: Proposal: spawn robot from topic

ValerioMa, 2020-12-09:
---
Create the robot model from topic
---

Since the parameter server is no more used in ros2, the robot description is now published as a string message. Tools like rviz2 can subscribe to this message and load the robot definition. 
I wanted to get a feedback if spawning a robot given the name of the topic could be a new feature to add inside *create.cpp* in the `ros_ign_gazebo` package.

Currently the model can be loaded from different source (param, file...) and I would propose to add a new argument that let user specify the name of the topic from which to load the robot model.

It would be sufficient to run
```sh
rosrun ros_ign_gazebo create --topic /robot_description
```

I am starting to port some simulation in ros2 and ignition gazebo citadel and this feature could make my actual code cleaner, but I am not sure if maybe I am missing something.

chapulina, 2020-12-09:
Thanks for bringing this up, I think that using the topic is fine, but I'm still wondering about the long-term recommended approach will be. 

`gazebo_ros_pkgs` has gone the opposite direction from `ros_ign`. It started supporting just a topic, like RViz2, and now there's some push to also support parameters: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1039#issuecomment-582126233. On `ros_ign` we currently support a parameter passed to the `create` node, but no topic yet.

It will be unfortunate if every ROS application interested in robot descriptions needs to support both topic and param.

ValerioMa, 2020-12-09:
Yes I agree. At the moment it is unclear the way to go and in long term this can lead to a set of  packages with different standard that will require some avoidable effort  to work together...as you mentioned this will be *unfortunate*.

clalancette, 2020-12-09:
My thinking here at the time I rewrote `robot_state_publisher` for ROS 2 was that a topic is better, since it is more dynamic.  That is, there can be one authority for the robot description in the system, and if you want to update it, it can publish the changes to everything downstream.

However, that is an aspirational goal.  Currently the system doesn't quite support that, and https://github.com/ros-simulation/gazebo_ros_pkgs/pull/1039 was a practical compromise to make things work today.

I think it would be much nicer to have `ros_ign` support a topic, but before doing that someone should also investigate how that would work when using ROS 2 launch files and simulation together.

chapulina, 2020-12-09:
Thanks for the input, @clalancette.

So @ValerioMa , if you're interested on working on an implementation, I think your suggestion of adding a `--topic` option would be fine.

> investigate how that would work when using ROS 2 launch files and simulation together.

We should at the very least add a working demo launch file to `ros_ign_gazebo_demos`, which I assume would use `robot_state_publisher` to publish the description to a topic. Ideally, we'd also add a test case to `ros_ign_gazebo` for this feature, but I wouldn't push too hard on that considering we currently don't have any tests for that package yet.

ValerioMa, 2020-12-11:
Yes by using a the publisher-subscriber paradigm to get the robot description I agree with @clalancette. With reference to the create.cpp it will be possible with limited effort to recreate a robot each time we get a new message on the topic. Obviously the creation node has to be kept alive for the whole simulation and we will need to subscribe to the `/pose/info` gazebo topic to be able to re-spawn the robot in the same position.
I am thinking if this could be useful and has some application. Maybe if we want to simulate reconfigurable modular robot abstracting away the physics involved in the reconfiguration procedure

vatanaksoytezer, 2021-08-18:
@chapulina I see this is already done and can be closed?

chapulina, 2021-08-18:
Yup, done in #128

PR #128 (2020-12-22): Add topic flag to create robot 
Fixes #127 

## Proposed Changes

  - add new --topic parameter that enables to specify the name of the topic where to read the robot description
  - add a demo launch file that loads the robot description on the `\robot_description` topic by leveraging the `robot_state_publisher ` and then use the newly defined flag to spawn the model in gazebo and additionally the launch starts rviz2 loading the robot description
  -  check `rclcpp::ok()` while server time to download resources. In case of failure the program will return -1
closed 2021-08-18 by chapulina