gazebosim/ros_gz #130: Melodic upgrade

peci1, 2020-12-29:
It seems to me that the binary packages `ros-melodic-ros-ign-*` from packages.osrfoundation.org started to target Dome instead of Blueprint after the changes from mid-December.

In my opinion, this is a pretty disruptive change - running `apt-get dist-upgrade` would install a newer version of Ignition Gazebo if I wouldn't have noticed. And then a lot of surprises would come. This really isn't a pleasant user experience, and it doesn't make me feel safe about using the ROS-Ign bridge in the future.

I think a better process needs to be defined. My idea is that it would be helpful if the binary packages would contain both the ROS and Ignition version in their name - e.g. ros-melodic-ros-ign-blueprint-bridge, ros-melodic-ros-ign-bridge-blueprint or ros-melodic-ign-blueprint-ros-ign-bridge. This way, no unintentional upgrades would happen. And transitional packages not bound to particular Ignition releases by their name could be created for the meantime - until people adopt the new names. Or they could be there forever, just pointing to one of the version-bound packages.

chapulina, 2020-12-29:
> It seems to me that the binary packages ros-melodic-ros-ign-* from packages.osrfoundation.org started to target Dome instead of Blueprint after the changes from mid-December.

Yes, they have, shortly after https://github.com/ignitionrobotics/ros_ign/pull/126 was merged. We planned to make an announcement ahead of time, but with the holidays I think we let it slip. Sorry about that.

> In my opinion, this is a pretty disruptive change

We think so too, and we don't plan on making this kind of change in the future. Melodic was a special case because of the SubT competition, which will use Dome now that Blueprint is going EOL. Noetic and Foxy packages will always target Citadel, and subsequent ROS 2 distros will stick to the version defined in [REP-2000](https://ros.org/reps/rep-2000.html).

> My idea is that it would be helpful if the binary packages would contain both the ROS and Ignition version in their name 

We discussed this idea in https://github.com/ignitionrobotics/ros_ign/issues/76 and decided against it because we believe it would make migration and maintenance harder for users.

peci1, 2020-12-29:
Okay, thanks for explaining. If future versions will always target only a single ROS distro, then this should not happen and adding Ignition collection name to packages is really not needed.

closed 2020-12-29 by peci1