gazebosim/ros_gz #131: [ROS1] No TF published from Ignition with DiffDrive plugin

doisyg, 2021-01-12:
As opposite to gazebo classic, no tf from `frame_id` to `child_frame_id`is published with ignition when using the DiffDrive plugin.
What would be the best implementation strategy ?
Adding in ros_ign_bridge a tf publisher when converting Odom messages ?
Or adding in Ignition DiffDrive plugin the publication of the newly added https://github.com/ignitionrobotics/ros_ign/commit/7b06c210379680bfa96f3e1d7794708f0e97007e  `ignition.msgs.Pose_V` that ros_ign_bridge will then convert to TF ?  Btw, these new commits were only pushed to the `melodic` branch, would it be possible to port them to `noetic` (as I am working with noetic) ? @chapulina

doisyg, 2021-01-12:
In fact, very easy to solve since there is a ign to ros Pose_V to tf2 conversion working in noetic, see https://github.com/ignitionrobotics/ign-gazebo/pull/548

AndreasAZiegler, 2021-03-30:
I get a
```
 Failed to create a bridge for topic [/rcv_transform] with ROS2 type [tf2_msgs/msg/TFMessage] and Ignition Transport type [ignition.msgs.Pose_V]
 ```
 when I try to start the bridge with

```
ros2 run ros_ign_bridge parameter_bridge /rcv_transform@tf2_msgs/msg/TFMessage@ignition.msgs.Pose_V
```

with

```
ign topic -i -t /rcv_transform
Publishers [Address, Message Type]:
  tcp://172.17.0.1:38643, ignition.msgs.Pose_V
```

doisyg, 2021-03-30:
Works for me with everything running on the same host.
Is the bridge working with other topics ? Which ros_ign version/branch are you running ?
I am closing this ticket a it should have after the PR was merged. Please open another one

closed 2021-03-30 by doisyg