gazebosim/ros_gz #133: Launch file may start wrong ign-gazebo version

chapulina, 2021-01-26:
`ign_gazebo.launch` starts Ignition Gazebo with the `ign gazebo` command line command:

https://github.com/ignitionrobotics/ros_ign/blob/a4bc25fde117f9ce7279c83dc419bfb0eb3983b9/ros_ign_gazebo/launch/ign_gazebo.launch.py#L34

https://github.com/ignitionrobotics/ros_ign/blob/968c31586bbe9964a3788d5970a90f88dfc3cb25/ros_ign_gazebo/scripts/ign_gazebo#L16

The problem is that, if a user has more than one `ign-gazebo` version installed, that will launch the latest version, which may not necessarily be what the ROS user wants. For example, if I have `ros-foxy-ros-ign` (which uses Citadel, i.e. `ign-gazebo3`) and Ignition Dome (i.e. `ign-gazebo4`) installed, the launch file will start a version of Ignition that may not be compatible with the rest of the things being launched.

The command line tool accepts the `--force-version` argument to specify what version to launch. But that is quite strict at the moment, and requires passing the fully qualified version, including minor and patch numbers (see https://github.com/ignitionrobotics/ign-tools/issues/36). This makes it complicated to hardcode the version on the launch file.

caioaamaral, 2021-03-21:
@chapulina I'm addressing `--force-version` here at ignitionrobotics/ign-tools#44 and currently writing a bugfix for this one.

chapulina, 2021-03-22:
Thanks, @caioaamaral !

ahcorde, 2022-03-25:
Added a new launch argument https://github.com/ignitionrobotics/ros_ign/pull/226

closed 2022-03-28 by chapulina