gazebosim/ros_gz #147: [ros2] Different interpretation of xacro/urdf files between **igniton-gazebo3** and  **igniton-gazebo4/5**

doisyg, 2021-04-02:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build: igniton-gazebo3 igniton-gazebo4 igniton-gazebo5

## Description
Don't know if I should report here or there: https://github.com/ignitionrobotics/ign-gazebo
I have a ROS / Ignition robot simulation working well with `citadel`, I did few tests with the newly released `edifice` and run into an issue on how my xacro/files are interpreted by gazebo when spawning with `ros_ign_gazebo create`.
More specifically on how a lidar sensor is positioned by gazebo.
When interpreting the following xacro/urdf file:

```
<?xml version="1.0" ?>
<robot name="will_be_ignored" xmlns:xacro="http://www.ros.org/wiki/xacro">

<gazebo>
  <plugin filename="ignition-gazebo-sensors-system" name="ignition::gazebo::systems::Sensors">
    <render_engine>ogre2</render_engine>
  </plugin>
</gazebo>

<link name="base_footprint">
  <visual>
    <geometry>
      <sphere radius="0.5"/>
    </geometry>
  </visual>
  <collision>
    <geometry>
      <sphere radius="0.5"/>
    </geometry>
  </collision>
  <inertial>
    <mass value="1"/>
  <inertia ixx="1" ixy="0.0" ixz="0.0" iyy="1" iyz="0.0" izz="1"/>
  </inertial>
</link>

<link name="link">
  <visual>
    <geometry>
      <sphere radius="1.0"/>
    </geometry>
  </visual>
  <collision>
    <geometry>
      <sphere radius="1.0"/>
    </geometry>
  </collision>
  <inertial>
    <mass value="1"/>
    <inertia ixx="1" ixy="0.0" ixz="0.0" iyy="1" iyz="0.0" izz="1"/>
  </inertial>
</link>

<gazebo reference="link" >
  <sensor name="link" type="gpu_lidar">
    <topic>scan</topic>
    <update_rate>30</update_rate>
    <lidar>
      <scan>
        <horizontal>
          <samples>2800</samples>
          <resolution>1</resolution>
          <min_angle>-2.09</min_angle>
          <max_angle>2.09</max_angle>
        </horizontal>
        <vertical>
          <samples>1</samples>
          <resolution>0.1</resolution>
          <min_angle>0.0</min_angle>
[... 357 lines cut ...]
                <resolution>0.10000000000000001</resolution>
                <min_angle>0</min_angle>
                <max_angle>0</max_angle>
              </vertical>
            </scan>
            <range>
              <min>0.050000000000000003</min>
              <max>30</max>
              <resolution>0.01</resolution>
            </range>
          </lidar>
          <visualize>1</visualize>
          <alwaysOn>1</alwaysOn>
          <plugin name='__default__' filename='__default__'/>
        </sensor>
      </link>
      <static>0</static>
      <plugin name='ignition::gazebo::systems::Sensors' filename='ignition-gazebo-sensors-system'>
        <render_engine>ogre2</render_engine>
      </plugin>
      <pose>1.2 0 0.5 0 -0 -2.88319</pose>
    </model>
  </world>
</sdf>
```

chapulina, 2021-04-05:
I suspect this may be related to changes from libSDFormat 9 (Citadel) to libSDFormat 10 (Dome), because that's where the URDF gets parsed into SDF. In fact, a quicker way of reproducing this can be achieved with the `ign sdf` tool:

Citadel has the sensor pose:

<details>
  <summary>ign sdf --force-version 9.5.0 -p /tmp/urdf</summary>
  
```
<sdf version='1.7'>
  <model name='will_be_ignored'>
    <link name='base_footprint'>
      <inertial>
        <pose>0 0 1.5 0 -0 0</pose>
        <mass>2</mass>
        <inertia>
          <ixx>6.5</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>6.5</iyy>
          <iyz>0</iyz>
          <izz>2</izz>
        </inertia>
      </inertial>
      <collision name='base_footprint_collision'>
        <pose>0 0 0 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>0.5</radius>
          </sphere>
        </geometry>
      </collision>
      <collision name='base_footprint_fixed_joint_lump__link_collision_1'>
        <pose>0 0 3 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>1</radius>
          </sphere>
        </geometry>
        <surface>
          <contact>
            <ode/>
          </contact>
          <friction>
            <ode/>
          </friction>
        </surface>
      </collision>
      <visual name='base_footprint_visual'>
        <pose>0 0 0 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>0.5</radius>
          </sphere>
        </geometry>
      </visual>
      <visual name='base_footprint_fixed_joint_lump__link_visual_1'>
        <pose>0 0 3 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>1</radius>
          </sphere>
        </geometry>
      </visual>
      <gravity>1</gravity>
      <velocity_decay/>
      <sensor name='link' type='gpu_lidar'>
        <topic>scan</topic>
        <update_rate>30</update_rate>
        <lidar>
          <scan>
            <horizontal>
              <samples>2800</samples>
              <resolution>1</resolution>
              <min_angle>-2.09</min_angle>
              <max_angle>2.09</max_angle>
            </horizontal>
            <vertical>
              <samples>1</samples>
              <resolution>0.1</resolution>
              <min_angle>0</min_angle>
              <max_angle>0</max_angle>
            </vertical>
          </scan>
          <range>
            <min>0.05</min>
            <max>30</max>
            <resolution>0.01</resolution>
          </range>
        </lidar>
        <visualize>1</visualize>
        <pose>0 0 3 0 -0 0</pose>
        <alwaysOn>1</alwaysOn>
      </sensor>
    </link>
    <static>0</static>
    <plugin name='ignition::gazebo::systems::Sensors' filename='ignition-gazebo-sensors-system'>
      <render_engine>ogre2</render_engine>
    </plugin>
  </model>
</sdf>
```

</details>

Dome doesn't:

<details>
  <summary>ign sdf --force-version 10.3.0 -p /tmp/urdf</summary>
  
```
<sdf version='1.7'>
  <model name='will_be_ignored'>
    <link name='base_footprint'>
      <inertial>
        <pose>0 0 1.5 0 -0 0</pose>
        <mass>2</mass>
        <inertia>
          <ixx>6.5</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>6.5</iyy>
          <iyz>0</iyz>
          <izz>2</izz>
        </inertia>
      </inertial>
      <collision name='base_footprint_collision'>
        <pose>0 0 0 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>0.5</radius>
          </sphere>
        </geometry>
      </collision>
      <collision name='base_footprint_fixed_joint_lump__link_collision_1'>
        <pose>0 0 3 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>1</radius>
          </sphere>
        </geometry>
        <surface>
          <contact>
            <ode/>
          </contact>
          <friction>
            <ode/>
          </friction>
        </surface>
      </collision>
      <visual name='base_footprint_visual'>
        <pose>0 0 0 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>0.5</radius>
          </sphere>
        </geometry>
      </visual>
      <visual name='base_footprint_fixed_joint_lump__link_visual_1'>
        <pose>0 0 3 0 -0 0</pose>
        <geometry>
          <sphere>
            <radius>1</radius>
          </sphere>
        </geometry>
      </visual>
      <gravity>1</gravity>
      <velocity_decay/>
      <sensor name='link' type='gpu_lidar'>
        <topic>scan</topic>
        <update_rate>30</update_rate>
        <lidar>
          <scan>
            <horizontal>
              <samples>2800</samples>
              <resolution>1</resolution>
              <min_angle>-2.09</min_angle>
              <max_angle>2.09</max_angle>
            </horizontal>
            <vertical>
              <samples>1</samples>
              <resolution>0.1</resolution>
              <min_angle>0</min_angle>
              <max_angle>0</max_angle>
            </vertical>
          </scan>
          <range>
            <min>0.05</min>
            <max>30</max>
            <resolution>0.01</resolution>
          </range>
        </lidar>
        <visualize>1</visualize>
        <alwaysOn>1</alwaysOn>
      </sensor>
    </link>
    <static>0</static>
    <plugin name='ignition::gazebo::systems::Sensors' filename='ignition-gazebo-sensors-system'>
      <render_engine>ogre2</render_engine>
    </plugin>
  </model>
</sdf>
```

</details>

Looking through [SDFormat's changelog](https://github.com/osrf/sdformat/blob/main/Changelog.md), nothing stood out as being the culprit though.

scpeters, 2021-04-05:
I believe this is connected to fixed-joint reduction with the libsdformat parser_urdf.cc file

chapulina, 2021-04-06:
I think this was fixed by https://github.com/osrf/sdformat/pull/525

doisyg, 2021-04-06:
Hopefully. It looks like it is related. Will test once merged or released in sdf11

doisyg, 2021-04-10:
Confirmed fixed in dome/sdf10, still need a merge in edifice/sdf11

chapulina, 2021-05-10:
The SDFormat fix has been forward-ported to Edifice. I'm closing this. ROS users will get the updated versions as the fix is released and ported to packages.ros.org.