gazebosim/ros_gz #157: [ros2] Enable ROS (or system) time header stamps in ROS Bridge

vatanaksoytezer, 2021-05-27:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Latest source of `ros_ign`, commit: Latest-fff2dbf4801cbe2116defb817f8a122536bb79d6
* Ignition: Edifice (Binary)
* ROS2: Foxy

## Description
* Expected behavior: All the messages that are bridged from ignition to ros2 are stamped with either real time or sim time. Having an option to use the bridge with ros (or system) time. Being able to only pull data with sim or real time is problematic due to time disagreements between ros nodes and ignition for some packages such as moveit2. 
* Actual behavior: There is only sim or real time available stamps on ignition messages that are communicated through `ros_ign_bridge`

## Steps to reproduce
I currently use https://github.com/vatanaksoytezer/ros_ign/pull/1 to work around this issue. I didn't bothered to open up a PR since this wouldn't get merged and breaks the sim time. But I would love to contribute if you point me out some ways through a parameter to add this functionality. I think I could also create and publish clock messages from ros2 to ignition, but that seems unintuitive and I am not even sure that would work.

DLu, 2021-06-08:
Thoughts @caguero ?

chapulina, 2021-06-15:
> Being able to only pull data with sim or real time is problematic due to time disagreements between ros nodes and ignition for some packages such as moveit2.

I may be mistaken, but shouldn't MoveIt be using sim time in this situation for consistency?

> Having an option to use the bridge with ros (or system) time.

Using system time with simulation seems unreliable to me. Simulation rarely runs exactly at real time, and depending on how fast or slow the simulation is running, you may end up with various strange behaviours.

> But I would love to contribute if you point me out some ways through a parameter to add this functionality.

But if this is really the feature that you need, maybe an environment variable like `ROS_IGN_TIME=system` could be used to switch the time source?

azeey, 2024-11-07:
This seems similar to https://github.com/gazebosim/ros_gz/issues/546 which has been fixed by https://github.com/gazebosim/ros_gz/pull/562

closed 2024-11-07 by azeey