gazebosim/ros_gz #162: Status of Foxy branch

mjcarroll, 2021-06-10:
In the process of moving over to github actions in (#159, #160, #161) and merging in some open pull requests (#158). I noticed that the `foxy` branch doesn't seem to actually be used.

The `foxy/distribution.yaml` confirms this: https://github.com/ros/rosdistro/blob/master/foxy/distribution.yaml#L3509-L3530

The release `tracks.yaml` also indicates this: https://github.com/ros2-gbp/ros_ign-release/blob/master/tracks.yaml#L15

Can the `foxy` branch be safely removed at this point?

chapulina, 2021-06-14:
The `foxy` branch was created as part of https://github.com/ignitionrobotics/ros_ign/pull/150. Due to a limitation in Bloom (see #517), we need to keep separate branches for `foxy` and `rolling` / `galactic` in order to release using different Ignition versions.

> The foxy/distribution.yaml confirms this

Good point, something to fix next time we make a `foxy` release

chapulina, 2021-07-20:
> something to fix next time we make a foxy release

https://github.com/ros/rosdistro/pull/30274

closed 2021-08-11 by chapulina