gazebosim/ros_gz #173: Regression in ROS Noetic on armv8 and armhf

sloretz, 2021-08-11:
@mjcarroll FYI

`ros_ign_bridge` is no longer building on Ubuntu Focal armhf and Debian Buster armv8 jobs due to ignitionrobotics/ros_ign#137

* http://repositories.ros.org/status_page/ros_noetic_ufhf.html?q=REGRESSION
* http://repositories.ros.org/status_page/ros_noetic_dbv8.html?q=REGRESSION

On these platforms it [appears to find ign-msgs5](https://build.ros.org/job/Nbin_dbv8_dBv8__ros_ign_bridge__debian_buster_arm64__binary/13/console).

<details><summary>Compiler error failure to find OccupancyGrid message</summary>

```
00:20:32.201 [  5%] Building CXX object CMakeFiles/ros_ign_bridge_lib.dir/src/convert.cpp.o
00:20:32.287 /usr/lib/ccache/c++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"ros_ign_bridge\" -Dros_ign_bridge_lib_EXPORTS -I/tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/include -I/opt/ros/noetic/include -I/opt/ros/noetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/include/ignition/msgs5 -isystem /usr/include/ignition/math6 -isystem /usr/include/ignition/transport8 -isystem /usr/include/ignition/cmake2 -isystem /usr/include/uuid  -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC   -Wall -Wextra -I/usr/include/uuid -std=gnu++17 -o CMakeFiles/ros_ign_bridge_lib.dir/src/convert.cpp.o -c /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp
00:20:32.294 In file included from /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:19:
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/include/ros_ign_bridge/convert.hpp:310:3: error: ‘ignition::msgs::OccupancyGrid’ has not been declared
00:20:55.500    ignition::msgs::OccupancyGrid & ign_msg);
00:20:55.500    ^~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/include/ros_ign_bridge/convert.hpp:315:25: error: ‘OccupancyGrid’ in namespace ‘ignition::msgs’ does not name a type
00:20:55.500    const ignition::msgs::OccupancyGrid& ign_msg,
00:20:55.500                          ^~~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:541:3: error: ‘ignition::msgs::OccupancyGrid’ has not been declared
00:20:55.500    ignition::msgs::OccupancyGrid & ign_msg)
00:20:55.500    ^~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp: In function ‘void ros_ign_bridge::convert_ros_to_ign(const ROS_T&, IGN_T&) [with ROS_T = nav_msgs::OccupancyGrid_<std::allocator<void> >; IGN_T = int]’:
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:543:48: error: request for member ‘mutable_header’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    convert_ros_to_ign(ros_msg.header, (*ign_msg.mutable_header()));
00:20:55.500                                                 ^~~~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:545:11: error: request for member ‘mutable_info’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    ign_msg.mutable_info()->mutable_map_load_time()->set_sec(
00:20:55.500            ^~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:547:11: error: request for member ‘mutable_info’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    ign_msg.mutable_info()->mutable_map_load_time()->set_nsec(
00:20:55.500            ^~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:550:11: error: request for member ‘mutable_info’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    ign_msg.mutable_info()->set_resolution(
00:20:55.500            ^~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:552:11: error: request for member ‘mutable_info’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    ign_msg.mutable_info()->set_width(
00:20:55.500            ^~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:554:11: error: request for member ‘mutable_info’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    ign_msg.mutable_info()->set_height(
00:20:55.500            ^~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:558:17: error: request for member ‘mutable_info’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500        (*ign_msg.mutable_info()->mutable_origin()));
00:20:55.500                  ^~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:560:11: error: request for member ‘set_data’ in ‘ign_msg’, which is of non-class type ‘int’
00:20:55.500    ign_msg.set_data(&ros_msg.data[0], ros_msg.data.size());
00:20:55.500            ^~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp: At global scope:
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:566:25: error: ‘OccupancyGrid’ in namespace ‘ignition::msgs’ does not name a type
00:20:55.500    const ignition::msgs::OccupancyGrid & ign_msg,
00:20:55.500                          ^~~~~~~~~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp: In function ‘void ros_ign_bridge::convert_ign_to_ros(const IGN_T&, ROS_T&) [with ROS_T = nav_msgs::OccupancyGrid_<std::allocator<void> >; IGN_T = int]’:
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:569:30: error: request for member ‘header’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    convert_ign_to_ros(ign_msg.header(), ros_msg.header);
00:20:55.500                               ^~~~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:571:44: error: request for member ‘info’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    ros_msg.info.map_load_time.sec = ign_msg.info().map_load_time().sec();
00:20:55.500                                             ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:572:45: error: request for member ‘info’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    ros_msg.info.map_load_time.nsec = ign_msg.info().map_load_time().nsec();
00:20:55.500                                              ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:573:37: error: request for member ‘info’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    ros_msg.info.resolution = ign_msg.info().resolution();
00:20:55.500                                      ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:574:32: error: request for member ‘info’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    ros_msg.info.width = ign_msg.info().width();
00:20:55.500                                 ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:575:33: error: request for member ‘info’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    ros_msg.info.height = ign_msg.info().height();
00:20:55.500                                  ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:577:30: error: request for member ‘info’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    convert_ign_to_ros(ign_msg.info().origin(), ros_msg.info.origin);
00:20:55.500                               ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:579:31: error: request for member ‘data’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    ros_msg.data.resize(ign_msg.data().size());
00:20:55.500                                ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:580:36: error: request for member ‘data’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    memcpy(&ros_msg.data[0], ign_msg.data().c_str(), ign_msg.data().size());
00:20:55.500                                     ^~~~
00:20:55.500 /tmp/binarydeb/ros-noetic-ros-ign-bridge-0.111.1/src/convert.cpp:580:60: error: request for member ‘data’ in ‘ign_msg’, which is of non-class type ‘const int’
00:20:55.500    memcpy(&ros_msg.data[0], ign_msg.data().c_str(), ign_msg.data().size());
00:20:55.500                                                             ^~~~
[... cut ...] bridge_lib.dir/src/convert.cpp.o] Error 1
```
</details>

Here are a few options to consider

1. Revert #137 and release a new version
2. Revert #137 temporarily, backport ignitionrobotics/ign-msgs#138 to Blueprint and Citadel, then release this feature again
3. [Remove Blueprint and Citadel support](https://github.com/ignitionrobotics/ros_ign/blob/3af3d38ada0e479313a433e24b0538581a6caaa1/ros_ign_bridge/package.xml#L23-26) and blacklist this package on the platforms that don't have Dome support.

chapulina, 2021-08-12:
> backport https://github.com/ignitionrobotics/ign-msgs/pull/138 to Blueprint and Citadel

Blueprint is EOL

It was backported to Citadel in https://github.com/ignitionrobotics/ign-msgs/pull/143 and released with 5.7.0: 

https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs5/Changelog.md#ignition-msgs-570-2021-03-17

> Remove Blueprint and Citadel support and blacklist this package on the platforms that don't have Dome support.

Note that the official version used by the ROS buildfarm for Noetic is Citadel, so we default to that on that branch:

https://github.com/ignitionrobotics/ros_ign/blob/noetic/ros_ign_bridge/package.xml

---

My proposed fix is to fix the buster armhf release and the upload of focal armhf to ROS:

### Buster armhf

It looks like the 5.7.0 release had a test failure on `buster:armhf`:

https://build.osrfoundation.org/job/ign-msgs5-debbuilder/120/console

```
[ RUN      ] UtilityTest.InitPointCloudPacked
/var/lib/jenkins/workspace/ign-msgs5-debbuilder/build/ignition-msgs-5.8.0/src/Utility_TEST.cc:689: Failure
Expected equality of these values:
  24u
    Which is: 24
  pc.point_step()
    Which is: 16
/var/lib/jenkins/workspace/ign-msgs5-debbuilder/build/ignition-msgs-5.8.0/src/Utility_TEST.cc:707: Failure
Expected equality of these values:
  16u
    Which is: 16
  pc.field(3).offset()
    Which is: 12
[  FAILED  ] UtilityTest.InitPointCloudPacked (6 ms)
```

### Focal armhf

For Focal armhf, the 5.7.0 release was successful:

https://build.osrfoundation.org/job/ign-msgs5-debbuilder/123/

But for some reason, 5.7.0 wasn't uploaded to the ROS packages:

http://packages.ros.org/ros2/ubuntu/pool/main/i/ignition-msgs5/

With this PR:

https://github.com/ros-infrastructure/reprepro-updater/pull/111

peci1, 2021-11-07:
I also can't find any arm64 binaries for melodic... Is it related?

chapulina, 2021-11-17:
The latest Ignition Msgs 5.8.1 has been uploaded to packages.ros.org with https://github.com/ros-infrastructure/reprepro-updater/pull/129. See:

* http://packages.ros.org/ros2/ubuntu/pool/main/i/ignition-msgs5/
* http://packages.ros.org/ros/ubuntu/pool/main/i/ignition-msgs5/

It's only missing Buster now, which should go in with:

https://github.com/ros-infrastructure/reprepro-updater/pull/132

After that's merged, we can trigger new Noetic releases.

> I also can't find any arm64 binaries for melodic... Is it related?

I'll defer to @j-rivero on this one, `melodic` is only released into packages.osrfoundation.org, so I suspect they can use the dependencies that are already there. I guess `arm64` is not being triggered? These docs should contain some clues:

https://github.com/ignitionrobotics/ros_ign/blob/melodic/RELEASING.md#releasing-into-httpspackagesosrfoundationorg

chapulina, 2021-12-30:
I believe we have all the necessary `ign-msgs` packages in packages.ros.org now:

* http://packages.ros.org/ros/ubuntu/pool/main/i/ignition-msgs5/

The new Noetic release should go through for all architectures:

* https://github.com/ros/rosdistro/pull/31643

Some `ros_ign` packages show as disabled on the status pages though:

* http://repositories.ros.org/status_page/ros_noetic_ufhf.html?q=ros_ign
* http://repositories.ros.org/status_page/ros_noetic_dbv8.html?q=ros_ign

Not sure what that means. Is there anything else we need to do to enable them, @sloretz ?

closed 2025-06-12 by ahcorde