gazebosim/ros_gz #174: Expose ignition::msgs::Contact through ROS 2 msgs

vatanaksoytezer, 2021-08-13:
Contact sensors are working nicely in Ignition through `ignition::msgs::Contact`. But they are not exposed through ros_ign. 

I don't mind adding another conversion to the `ros_ign_bridge`, but I am not %100 sure what ROS message type I should use. I couldn't fit this to any existing ROS2 message. I can of course define a custom message type or create a new message package (ala [gazebo_msgs](https://github.com/ros-simulation/gazebo_ros_pkgs)), but I just want this to be mergeable contribution, so this issue is to discuss what you would be happy to do. Let me know which option would be better to expose this and I will be happy to implement it @chapulina.

For reference `ignition::msgs::Contact` has type:

```
header {
  stamp {
    sec: 6
    nsec: 884000000
  }
}
contact {
  header {
    stamp {
      sec: 6
      nsec: 884000000
    }
  }
  collision1 {
    id: 96
  }
  collision2 {
    id: 7
  }
  position {
    x: -0.38790994410051866
    y: -0.21047000000004118
    z: -3.3862761483760551e-09
  }
  position {
    x: -0.38790994409679119
    y: -0.15047000000004118
    z: -3.4019649319816381e-09
  }
}

```

chapulina, 2021-08-13:
> define a custom message type or create a new message package (ala gazebo_msgs),

Ohh have you seen the ones recently added in https://github.com/ignitionrobotics/ros_ign/pull/152?

vatanaksoytezer, 2021-08-13:
> Ohh have you seen the ones recently added in #152?

Oh, does look neat! Will send a PR soon for bridge and hopefully close this after it gets merged! Thanks!

PR #175 (2021-08-18): Expose Contacts through ROS bridge

# 🎉 New feature

Closes #174

## Summary
The ultimate goal was to expose contact sensor messages through bridge. This adds a new message type to `ros_ign_interfaces`: JointWrench and adds six different types to `ros_ign_bridge`, namely:

- UInt32
- geometry_msgs/msgWrench
- ros_ign_interfaces/msg/JointWrench
- ros_ign_interfaces/msg/Entity
- ros_ign_interfaces/msg/Contact
- ros_ign_interfaces/msg/Contacts

I added UInt32 thinking I would need it but it turns out I don't but nevertheless it's just a new type for bridge 🎉 
## Test it
```bash
ign gazebo -r cont [... cut]
closed 2021-08-18 by vatanaksoytezer