gazebosim/ros_gz #177: Fatal Error when building ros_ign_bridge

ggiulian, 2021-08-28:
## Environment
* OS Version:  Ubuntu 20.04
* Source or binary build?
 Source: ROS2 Galactic and Ignition Edifice

    
<details>
  Following the documentation released here: https://github.com/ignitionrobotics/ros_ign/tree/ros2, the build fails with a fatal error when trying to build the ros_ign_bridge. In particular it cannot find the nav_msgs / msg / odometry.hpp file or directory. 
The file in question, however, appears to be present in the ROS2 Galactic workspace.
The error is present in the test_utils.hpp
```
# paste log here
```Starting >>> ros_ign_bridge
--- stderr: ros_ign_bridge                             
In file included from /home/ggiuliano/ros_ign_ws/src/ros_ign/ros_ign_bridge/test/subscribers/ign_subscriber.cpp:22:
/home/ggiuliano/ros_ign_ws/src/ros_ign/ros_ign_bridge/test/subscribers/../test_utils.hpp:42:10: fatal error: nav_msgs/msg/odometry.hpp: No such file or directory
   42 | #include <nav_msgs/msg/odometry.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/ggiuliano/ros_ign_ws/src/ros_ign/ros_ign_bridge/test/publishers/ign_publisher.cpp:25:
/home/ggiuliano/ros_ign_ws/src/ros_ign/ros_ign_bridge/test/publishers/../test_utils.hpp:42:10: fatal error: nav_msgs/msg/odometry.hpp: No such file or directory
   42 | #include <nav_msgs/msg/odometry.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/test_ign_subscriber.dir/build.make:63: CMakeFiles/test_ign_subscriber.dir/test/subscribers/ign_subscriber.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:327: CMakeFiles/test_ign_subscriber.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/test_ign_publisher.dir/build.make:63: CMakeFiles/test_ign_publisher.dir/test/publishers/ign_publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:300: CMakeFiles/test_ign_publisher.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ros_ign_bridge [0.84s, exited with code 2]

</details>

## Description
* Expected behavior: No build errors
* Actual behavior: nav_msgs/msg/odometry.hpp not found

chapulina, 2021-08-30:
Did you run `rosdep install -r --from-paths src -i -y --rosdistro galactic` as instructed in the README? It looks like you're missing `ros-galactic-nav-msgs`.

ggiulian, 2021-08-30:
[image]
As you can see, all rosdeps are installed

chapulina, 2021-08-30:
Our latest CI for Edifice + Galactic compiles and passes all tests:

https://github.com/ignitionrobotics/ros_ign/runs/3356214523?check_suite_focus=true

It also works for me locally.

So I think it's probably an issue with your workspace. Here are the steps followed by CI as a reference:

https://github.com/ignitionrobotics/ros_ign/blob/ros2/.github/workflows/build-and-test.sh

ggiulian, 2021-08-31:
Sorry, which workspace? ROS2 or ROS_IGN?

chapulina, 2021-08-31:
One of them :slightly_smiling_face: 

Did you install `nav-msgs` from source or binaries? That seems to be the problematic package.

ggiulian, 2021-08-31:
From source 
Also in the test_utils.hpp I tried to omit the `#include <nav_msgs/msg/odometry.hhp>` and I encountered the same build error but in the `sensor_msgs/msg/battery_state.hpp` package. So I think there are other problematic packages

chapulina, 2021-09-01:
> From source

My guess is that you probably have a bad combination of branches or sourced workspaces.

Since I'm unable to reproduce the issue and this also works on our automated testing, this seems to be a local problem with your setup. I'll close this issue and suggest you follow up at https://answers.gazebosim.org/.

closed 2021-09-01 by chapulina

matosinho, 2021-09-21:
The same thing just happened to me. The only difference is that I'm using ROS Foxy and Ignition Citadel.

matosinho, 2021-09-22:
The problem seems to be occuring during test compilation. For me, after adding `--cmake-args -DBUILD_TESTING=OFF
` in `colcon build`, the bridge compiles and seems to be working just fine. So, IMHO that might be something with CMake of the test, since the include from `convert.hpp` works and the one from `test_utils.hpp` don't. I still need to study more ament and take a deeper look in the `CMakeLists.txt` to see if I can understand why this happens. Also, I believe that most people use binary installation and for some reason, this problem doesn't occur there. CI is assuming binary installation as well, isn't it @chapulina? @ggiulian were you able to manage this situation in some other way?

ggiulian, 2021-09-22:
It worked for me, I can tell you the steps I took.
Before the build, in a new terminal:

1. `source /opt/ros/galactic/setup.bash`
2. `cd  ~/ros_ign_ws`
3. `source ~/edifice_ws/install/local_setup.bash`
4. `colcon build`

Following this steps, it works for me.
I agree that many use binary installation, so they don't notice this problem