gazebosim/ros_gz #190: ros_ign_gazebo create.cpp usage message incorrect

oldtopos, 2021-12-16:
and misleading

## Environment
* OS Version: Ubuntu 20.04
* Source build, ros2 branch

## Description
* Expected behavior: 
* 
* gflags::SetUsageMessage(
    R"(Usage: create -world [arg] [-file FILE] [-param PARAM] [-topic TOPIC]
                       [-string STRING] [-name NAME] [-x X] [-y Y] [-z Z]
                       [-R ROLL] [-P PITCH] [-Y YAW])");
* Actual behavior:   
* 
* gflags::SetUsageMessage(
    R"(Usage: create -world [arg] [-file FILE] [-param PARAM] [-topic TOPIC]
                       [-string STRING] [-name NAME] [-X X] [-Y Y] [-Z Z]
                       [-Roll ROLL] [-Pitch PITCH] [-Yaw YAW])");

chapulina, 2021-12-16:
Oops, that's here:

https://github.com/ignitionrobotics/ros_ign/blob/5d9cfef3d58948a77b962d231e281c88ea88e188/ros_ign_gazebo/src/create.cpp#L51

Aryaman22102002, 2022-01-12:
I have tried to fix this issue. Can someone please take a look at the pull request.  Link to pr: https://github.com/ignitionrobotics/ros_ign/pull/206

Aryaman22102002, 2022-01-14:
Does this issue needs to be resolved for other branches too?

chapulina, 2022-01-14:
> Does this issue needs to be resolved for other branches too?

Yeah I think so. If you could open a PR for `foxy`, we can then forward-port it to `galactic`. Thanks!

chapulina, 2022-01-14:
Thanks @Aryaman22102002

closed 2022-01-14 by chapulina