gazebosim/ros_gz #205: Allow exporting IGN_GAZEBO_RESOURCE_PATH, etc... in the package.xml file of a ROS package

AndPatr, 2022-01-08:
## Desired behavior
I am using ROS2 foxy and Ignition Gazebo to write/simulate a package for my robot.
On Gazebo Classic, I am able to avoid modifying manually my .bashrc by inserting an export tag <gazebo_ros> in my package.xml .
This way, when someone build my package, the robot files are automatically available and the package can be tested immediately without having to touch the .bashrc .
I would like to see something similar implemented for ros_ign (I have searched for it, but I did not find this feature).

## Implementation suggestion
Probably, something like [https://github.com/ros-simulation/gazebo_ros_pkgs/blob/foxy/gazebo_ros/scripts/gazebo_ros_paths.py](gazebo_ros_paths) would be enough.

chapulina, 2022-01-10:
I vaguely remember the `package.xml` approach for `gazebo_ros_pkgs` having some issues, but I'm not finding any tickets with information right now.

In any case, we've been using env hooks to export environment variables from other packages. One advantage that has over the `gazebo_ros_paths` approach is that it's easier to introspect when you source a workspace.

---

See an example here:

1. Create the [env hook file](https://github.com/chapulina/dolly/blob/299ba72af995e75d30e2bca658a1078bce301aee/dolly_ignition/env-hooks/dolly_ignition.dsv.in#L1-L2) exporting `IGN_GAZEBO_RESOURCE_PATH`
2. Call [`ament_environment_hooks`](https://github.com/chapulina/dolly/blob/299ba72af995e75d30e2bca658a1078bce301aee/dolly_ignition/CMakeLists.txt#L42)

And you're all set.

---

Unless there are downsides to the env hook approach, I'm inclined not to support yet another way of setting environment variables.

chapulina, 2022-01-25:
@AndPatr , I'll close this issue for now, but please comment here if you still see a need for it.

closed 2022-01-25 by chapulina