gazebosim/ros_gz #218: NavSat Ignition Gazebo ROS Bridge Update

jugpanther, 2022-02-22:
Hello, 

I am curious if anyone has been working on getting the NavSat/GPS parameter in Ignition Gazebo converted with the ign_ros bridge. Using Citadel I was able to get the NavSat Ignition Gazebo sensor message to work, but I didn't see the ROS message conversion available in the sensor list. Any tips on how I can get the GPS sensor message converted to ROS.

ahcorde, 2022-02-22:
Hi @jugpanther,

Just in case you want to go ahead and add this feature I drop here some tips:

 - Create the headers [here](https://github.com/ignitionrobotics/ros_ign/blob/melodic/ros_ign_bridge/include/ros_ign_bridge/convert.hpp) for the convertion functions IGN->ROS and ROS->IGN 
 - Implement [here](https://github.com/ignitionrobotics/ros_ign/blob/melodic/ros_ign_bridge/src/convert.cpp) the convertions 
 - Add [here](https://github.com/ignitionrobotics/ros_ign/blob/melodic/ros_ign_bridge/src/factories.hpp) the header for the factory 
 - Implement [here](https://github.com/ignitionrobotics/ros_ign/blob/melodic/ros_ign_bridge/src/factories.cpp) the factory method
 - And finally add an entry in the `get_factory_impl` in the factory.cpp file

Tips are the some for ROS or ROS 2. You should use the right APIs.

Add me as a reviewer if you create a PR

jugpanther, 2022-02-24:
@ahcorde, 

Thank you for the tips. I will see what I can do.

PR #224 (2022-04-22): Add support for bridging NavSat
# 🎉 New feature

Closes #218

## Summary
Adds two-way bridge for NavSat messages between ROS and IGN along with a demo launch file.

## Test it
Test IGN->ROS by running `roslaunch ros_ign_gazebo_demos navsat.launch`.
Test ROS->IGN by `roslaunch ros_ign_gazebo_demos navsat.launch` and `ign topic -t /navsat -e`
```
rostopic pub -r 1 /navsat sensor_msgs/NavSatFix "header:
  seq: 10
  stamp: {secs: 10, nsecs: 0}
  frame_id: 'back_to_ign'
status: {status: 0, service: 0}
latitude: 10.0
longitude: 10.0
altitude: 10.0
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0,  [... cut]
closed 2022-04-22 by chapulina