gazebosim/ros_gz #219: ros_ign in ROS 2 Rolling has not been released on Ubuntu Jammy

nuclearsandwich, 2022-02-23:
The ROS 2 Rolling Ridley distribution recently [migrated to Ubuntu 22.04 Jammy](https://github.com/ros/rosdistro/pull/32036) using the rosdistro migration tool.

This repository is one which failed to bloom during the migration because the ignition dependencies are not yet available in Ubuntu Jammy. Once the Jammy repositories and the rosdep database are updated it should be possible to release this repository in Rolling again by running bloom normally.

Packages which are not released in Rolling may not be automatically added to future stable ROS 2 distributions.

ruffsl, 2022-02-28:
With Ubuntu Jammy not listed as a supported platform for Gazebo V11, I'm guessing Nav2 will have to migrate it's system simulation tests to Ignition. So having `ros_ign` installable from Rolling for migrating Nav2's CI would be a nice to have.

chapulina, 2022-03-01:
These need to be done first:

- [ ] Release Ignition Fortress for Jammy into packages.osrfoundation.org: https://github.com/ignition-tooling/release-tools/issues/566
- [ ] Port those packages to packages.ros.org
    - similar to https://github.com/ros-infrastructure/reprepro-updater/pull/128
    - it should be easier with https://github.com/ros-infrastructure/reprepro-updater/pull/144)
- [ ] Add rosdep keys 
    - similar to https://github.com/ros/rosdistro/pull/31591

---

Let us know if you need any help while migrating Nav2 to Ignition, @ruffsl , we're happy to provide guidance.

ivanpauno, 2022-03-23:
@chapulina just checking, it seems that the first two steps were done (https://github.com/ros-infrastructure/reprepro-updater/pull/143), right?

Are only the rosdep keys missing?

chapulina, 2022-03-23:
https://github.com/ros-infrastructure/reprepro-updater/pull/143 seems to be only partial, it has only `ign-cmake` and `ign-math`.

We do have debs for all of Fortress available already, and I believe @j-rivero is planning to kick off the reprepro process in the coming days.

hardesh, 2022-04-22:
I think `ros_ign_gazebo` needs `std_msgs` as its dependency. I was just trying out the rolling branch on Ubuntu 22.04 and got a build failure. Package built properly after adding the dependency

chapulina, 2022-04-22:
> ros_ign_gazebo needs std_msgs as its dependency

* See https://github.com/ignitionrobotics/ros_ign/pull/242

chapulina, 2022-04-25:
It should be out on the next sync:

* https://github.com/ros/rosdistro/pull/32889
* https://github.com/ros/rosdistro/pull/32890

closed 2022-04-25 by chapulina