gazebosim/ros_gz #220: ROS2 galactic, ign fortress, find_package(geometry_msgs REQUIRED)

ymei, 2022-03-04:
I found that in order to compile (`colcon build`) successfully on ROS2, `ros_ign_gazebo/CMakeLists.txt` needs to be modified:

```
find_package(geometry_msgs REQUIRED)
ament_target_dependencies(create
  rclcpp
  ignition-math6
  geometry_msgs
)
```
Otherwise, it results in a linking error:
```
undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<std_msgs::msg::String_<std::allocator<void> > >()'
```

chapulina, 2022-03-04:
What branch are you on? And what ROS version are you using?

ymei, 2022-03-04:
branch: ros2
ROS2 version: galactic
ignition version: fortress

ymei, 2022-03-04:
The `galactic` branch has the same problem.  It needs the same modification to compile for me.

chapulina, 2022-03-04:
I just re-ran our CI and it compiled fine. I wonder what's different on your setup.

https://github.com/ignitionrobotics/ros_ign/runs/5428022992?check_suite_focus=true

ymei, 2022-03-04:
Any information I can provide to help debug this?  Please let me know.  I'm no expert.  I got the idea to add `geometry_msgs` to solve my problem from this page: https://answers.ros.org/question/315023/ros2-debugging-techniques/

chapulina, 2022-06-13:
`std::msgs` added to the `ros2` branch in https://github.com/gazebosim/ros_gz/pull/242

closed 2022-06-13 by chapulina