gazebosim/ros_gz #235: Memory Overflow on ROS2 Rolling

hardesh, 2022-04-07:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Source

Hi,
I'm facing issue while building `ros_ign_bridge` on Ubuntu 20 with ROS2 Rolling with Ignition Fortress. Memory on my PC is overflowing everytime i try to build the package. I tried building older commit which had a CI check. But faced the same issue with that commit. 

Specifications:
CPU: AMD Ryzen 7 3700x 8-core processor × 16 
RAM: 16GB

hardesh, 2022-04-07:
Closing this issue. It built fine after I closed the browser. This was weird

hardesh, 2022-04-07:
Hi,
My bad. Still facing issue with `ros2` branch. `galactic` branch is working fine

mjcarroll, 2022-04-07:
We had to split up some of the translation units in `galactic` at least to get them through the build farm.  The number of templates is starting to make compilation time/memory usage an issue.

There is a forward port to bring the `galactic` changes forward to `rolling` here: https://github.com/ignitionrobotics/ros_ign/pull/228, but we are still blocked on some Jammy/Rolling transition issues as described here: https://github.com/ignitionrobotics/ros_ign/issues/219

Feel free to give that #228 a try locally and see if that resolves the issue for you.

hardesh, 2022-04-07:
Hi @mjcarroll,
`galactic` branch is working fine on my PC. But it still take a ton of memory while building almost 80%

mjcarroll, 2022-04-08:
> But it still take a ton of memory while building almost 80%

This is somewhat expected given the volume of templates that we have.  You can also attempt restrict the number of parallel jobs using either `make -j1` if you are in the build directory or you can use `colcon build --executor parallel --parallel-workers 1`

closed 2022-04-08 by mjcarroll