gazebosim/ros_gz #248: Building Gazebo Ignition Fortress for ROS2 humble on Ubuntu Jammy

dfarning, 2022-04-29:
For the last several days I have been testing ROS Humble which was released for beta testing on April 25, 2022. The basic features are pretty stable.

I have also tested Gazebo Ignition Rolling on ROS2 Rolling. I was able to run several demos without problems.

Would it be possible to start building Gazebo Ignition Fortress for ROS2 HUmble?

chapulina, 2022-05-02:
* See https://github.com/ignitionrobotics/ros_ign/issues/219

We've bloomed `ros_ign` for Humble and Rolling last week. I just ran [these instructions](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html) in a Docker container and I'm able to install `ros-humble-ros-ign`.

I'm closing this, but please comment if something else is missing for you.

closed 2022-05-02 by chapulina

dfarning, 2022-05-02:
Thanks, I am now able to install it.