gazebosim/ros_gz #253: Examples 4 failure .

xlla, 2022-05-23:
## Environment
* OS Version: macOS 10.13.6
* Source or binary build?
  source ,
  ros_ign_bridge,  main,  3512c016ad05fa56d325ad5c2f9c0dfafcb962a5
       build with transport12, msg9, math7, gazebo7
  ros2, humble
  ign gazebo7, gui7

## Description
* Expected behavior: 
  like Examples described
* Actual behavior: 

<pre>
slim:ros_gz xlla$ ros2 service call /world/shapes/control ros_ign_interfaces/srv/ControlWorld "world_control: pause: false"
Traceback (most recent call last):
  File "/opt/ros2/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli', 'console_scripts', 'ros2')())
  File "/Users/xlla/ros2_foxy/build/ros2cli/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/Users/xlla/ros2_foxy/build/ros2service/ros2service/command/service.py", line 41, in main
    return extension.main(args=args)
  File "/Users/xlla/ros2_foxy/build/ros2service/ros2service/verb/call.py", line 58, in main
    return requester(
  File "/Users/xlla/ros2_foxy/build/ros2service/ros2service/verb/call.py", line 80, in requester
    values_dictionary = yaml.safe_load(values)
  File "/usr/local/lib/python3.9/site-packages/yaml/__init__.py", line 125, in safe_load
    return load(stream, SafeLoader)
  File "/usr/local/lib/python3.9/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
  File "/usr/local/lib/python3.9/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/local/lib/python3.9/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python3.9/site-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/local/lib/python3.9/site-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/local/lib/python3.9/site-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/local/lib/python3.9/site-packages/yaml/scanner.py", line 577, in fetch_value
    raise ScannerError(None, None,
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 1, column 21:
    world_control: pause: false
                        ^

</pre>

## Steps to reproduce
as Examples 4.

## Output
terminal 1
<pre>
slim:bridge_ws xlla$ ros2 run ros_ign_bridge parameter_bridge /world/shapes/control@ros_ign_interfaces/srv/ControlWorld
</pre>

terminal 2
<pre>
[Msg] Ignition Gazebo Server v7.0.0~pre1
[... 202 lines cut ...]
    node = self.get_single_node()
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.9/site-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/local/lib/python3.9/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python3.9/site-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/local/lib/python3.9/site-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/local/lib/python3.9/site-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/local/lib/python3.9/site-packages/yaml/scanner.py", line 577, in fetch_value
    raise ScannerError(None, None,
yaml.scanner.ScannerError: mapping values are not allowed here
  in "<unicode string>", line 1, column 21:
    world_control: pause: false
                        ^

</pre>

ahcorde, 2022-05-24:
Hi @xlla, 

do you mind to try this fix ? https://github.com/gazebosim/ros_gz/pull/254

xlla, 2022-06-08:
hi @ahcorde ,  it works again using #254 . thanks you!
```

 ros2 service call /world/shapes/control ros_ign_interfaces/srv/ControlWorld "{world_control: {pause: false}}"
requester: making request: ros_ign_interfaces.srv.ControlWorld_Request(world_control=ros_ign_interfaces.msg.WorldControl(pause=False, step=False, multi_step=0, reset=ros_ign_interfaces.msg.WorldReset(all=False, time_only=False, model_only=False), seed=0, run_to_sim_time=builtin_interfaces.msg.Time(sec=0, nanosec=0)))

response:
ros_ign_interfaces.srv.ControlWorld_Response(success=True)

```