gazebosim/ros_gz #266: Bridge for service calls on galactic `SetEntityPose`

marc-wittwer, 2022-06-27:
I'm interested in accessing the `SetEntityPose` service via the ros_bridge.
I saw that ControlWorld service is available for rolling. Is the 

I'm currently on ROS galactic. Is it possible to also bridge service calls in galactic? 
How can I help to make it available in galactic?

I just need a way to set the pose of an entity in gazebo programmatically. I found that using the service call via the terminal is very slow, thus I want to call the service using the ign transport directly.

marc-wittwer, 2022-06-30:
I managed to call the set_pose service using C++.

Here is my script: 
https://github.com/gazebosim/gz-transport/issues/323#issuecomment-1171263362

chapulina, 2022-07-07:
In theory, this can be backported to galactic:

* https://github.com/gazebosim/ros_gz/pull/211

But it may be a lot of work, considering that the code on the `ros2` branch has significantly diverged from `galactic`. And it also may not be worth it, since Galactic has only ~4 months left of support.