gazebosim/ros_gz #269: ros_ign_bridge

oKermorgant, 2022-06-29:
Hi, 

Using Galactic + Fortress here.

## Desired behavior
The current `parameter_bridge` uses Ignition topic names on the ROS side, as only one topic is included in the `<topic@ROS2_type@Ign_type>` syntax.

For some topics, this leads to an invalid ROS topic name. This is the case for e.g. joint commands where the Ignition topic is `/model/<model_name>/joint/<joint_name>/<joint_index>/cmd_*`, but `<joint_index>` being an integer this makes it invalid for ROS.

Remapping should be a solution but even running the `parameter_bridge` with some remappings, the original topic name is still considered while creating the publisher/subscriber, which raises an error:

`parameter_bridge /model/baxter/joint/right_s0/0/cmd_pos@std_msgs/msg/Float64]ignition.msgs.Double --ros-args -r /model/baxter/joint/right_s0/0/cmd_pos:=right_s0_cmd_pos`

- `Couldn't parse remap rule: '-r /model/baxter/joint/right_s0/0/cmd_pos:=right_s0_cmd_pos'`

## Alternatives considered
Currently an alternative is to use a custom bridge that does not use the same name for Ignition topics and ROS ones, for these particular topics. 

## Implementation suggestion

1. A backward-compatible solution could be to allow an optional additional parameter in the command line. 
`<topic@ROS2_type@Ign_type|ros_topic>` (arbitrary `|` separator)
If this parameter is not here, the fallback to `ros_topic = topic` is straightforward.
2. Another way is to allow ROS parameters named after Ignition topics, that would be set to the corresponding ROS topic, basically replacing `-r` by `-p` to bypass the remapping process:
`parameter_bridge /model/baxter/joint/right_s0/0/cmd_pos@std_msgs/msg/Float64]ignition.msgs.Double --ros-args -p /model/baxter/joint/right_s0/0/cmd_pos:=right_s0_cmd_pos`
This one is probably more tedious to write as this ROS parameter may or may not be defined.

mjcarroll, 2022-06-29:
I have added a mechanism for configuring the bridge with a yaml file, which allows you to directly set the mapping between Ignition and ROS topics without having to deal with the remapping rules.  See if this is a reasonable solution: https://github.com/gazebosim/ros_gz/pull/238

I opted for this rather than extending the current syntax because I found YAML to be more consistent to work with, and it allows users to check their configuration into version control.

oKermorgant, 2022-06-29:
Great! That is clearly doing the job.
How do one specify between ros->gz and gz->ros in the yaml file?

mjcarroll, 2022-06-29:
> How does one specify between ros->gz and gz->ros in the yaml file?

Ah, I see that wasn't included in the documentation, I will update that.  Without any other configuration, the default is bidirectional comms.  You can additionally set directions with the following:

```
direction: BIDIRECTIONAL
direction: IGN_TO_ROS
direction: ROS_TO_IGN
```

mjcarroll, 2022-06-29:
Let me know if #271 is a sufficient improvement to the documentation to help future users.

oKermorgant, 2022-06-29:
Yep I think this covers all use cases now.

Python-based launch files could more or less easily build the `parameter_bridge` `@voodoo]` strings but a YAML file is much better.
I guess the `IGN_TO_ROS` will be changed to `GZ_TO_ROS` at some point. 

Anyway thanks for the pointer to this PR.

mjcarroll, 2022-06-29:
>  I guess the IGN_TO_ROS will be changed to GZ_TO_ROS at some point.

If I understand correctly, the `galactic` branch will not have the renaming applied, only `ros2` and forward.  @chapulina can you confirm?

chapulina, 2022-06-29:
Correct, the gz renaming will only apply from Rolling / Iron. 

We considered adding various `gz` aliases for Foxy and Galactic, so that users could start using `gz` with them if they wanted to future-proof. But considering that these 2 distributions have a relatively short life ahead of them, it's not worth the effort.

closed 2022-06-29 by mjcarroll