gazebosim/ros_gz #288: ros_ign_bridge test_launch_test_ros_subscriber.launch.py fails intermittently

andermi, 2022-08-19:
## Environment
* OS Version: Ubuntu 20.04
* Source: galactic, 94abe668a8e5ef8cfad9059f9e8ce125f12415c6
  
## Description
ros_ign_bridge test_launch_test_ros_subscriber.launch.py fails intermittently

## Steps to reproduce
1. clone, checkout, build
2. colcon test

## Output
```
test 3
[0.171s]       Start  3: test_launch_test_ros_subscriber.launch.py
[0.171s] 
[0.171s] 3: Test command: /usr/bin/python3 "-u" "/opt/ros/galactic/share/ament_cmake_test/cmake/run_test.py" "/home/anderson/wec/igngzb/openrobotics/workspace/build/ros_ign_bridge/test_results/ros_ign_bridge/test_launch_test_ros_subscriber.launch.py.xunit.xml" "--package-name" "ros_ign_bridge" "--output-file" "/home/anderson/wec/igngzb/openrobotics/workspace/build/ros_ign_bridge/launch_test/test_launch_test_ros_subscriber.launch.py.txt" "--command" "/usr/bin/python3" "-m" "launch_testing.launch_test" "/home/anderson/wec/igngzb/openrobotics/workspace/src/ros_ign/ros_ign_bridge/test/launch/test_ros_subscriber.launch.py" "gz_msgs_ver:=8.6.0" "--junit-xml=/home/anderson/wec/igngzb/openrobotics/workspace/build/ros_ign_bridge/test_results/ros_ign_bridge/test_launch_test_ros_subscriber.launch.py.xunit.xml" "--package-name=ros_ign_bridge"
[0.171s] 3: Test timeout computed to be: 200
[0.210s] 3: -- run_test.py: invoking following command in '/home/anderson/wec/igngzb/openrobotics/workspace/build/ros_ign_bridge':
[0.210s] 3:  - /usr/bin/python3 -m launch_testing.launch_test /home/anderson/wec/igngzb/openrobotics/workspace/src/ros_ign/ros_ign_bridge/test/launch/test_ros_subscriber.launch.py gz_msgs_ver:=8.6.0 --junit-xml=/home/anderson/wec/igngzb/openrobotics/workspace/build/ros_ign_bridge/test_results/ros_ign_bridge/test_launch_test_ros_subscriber.launch.py.xunit.xml --package-name=ros_ign_bridge
[0.425s] 3: [INFO] [launch]: All log files can be found below /home/anderson/.ros/log/2022-08-18-16-50-57-737194-Gladosu-3464716
[0.426s] 3: [INFO] [launch]: Default logging verbosity is set to INFO
[0.465s] 3: test_termination (ros_ign_bridge.ROSSubscriberTest) ... [INFO] [test_ign_publisher-1]: process started with pid [3464719]
[0.465s] 3: [INFO] [test_ros_subscriber-2]: process started with pid [3464721]
[0.465s] 3: [INFO] [parameter_bridge-3]: process started with pid [3464723]
[0.465s] 3: [INFO] [python3-4]: process started with pid [3464725]
[0.482s] 3: [test_ros_subscriber-2] [==========] Running 45 tests from 1 test suite.
[0.482s] 3: [test_ros_subscriber-2] [----------] Global test environment set-up.
[0.482s] 3: [test_ros_subscriber-2] [----------] 45 tests from ROSSubscriberTest
[0.482s] 3: [test_ros_subscriber-2] [ RUN      ] ROSSubscriberTest.vector3d_point
[0.517s] 3: [parameter_bridge-3] [INFO] [1660866657.901788223] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/vector3d_point (ignition.msgs.Vector3d) -> /vector3d_point (geometry_msgs/msg/Point)] (Lazy 0)
[0.518s] 3: [parameter_bridge-3] [INFO] [1660866657.902900337] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/pose_pose (ignition.msgs.Pose) -> /pose_pose (geometry_msgs/msg/Pose)] (Lazy 0)
[0.518s] 3: [parameter_bridge-3] [INFO] [1660866657.903379264] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/pose_posestamped (ignition.msgs.Pose) -> /pose_posestamped (geometry_msgs/msg/PoseStamped)] (Lazy 0)
[0.519s] 3: [parameter_bridge-3] [INFO] [1660866657.903830133] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/quaternion_quaternion (ignition.msgs.Quaternion) -> /quaternion_quaternion (geometry_msgs/msg/Quaternion)] (Lazy 0)
[0.519s] 3: [parameter_bridge-3] [INFO] [1660866657.904277476] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/pose_transform (ignition.msgs.Pose) -> /pose_transform (geometry_msgs/msg/Transform)] (Lazy 0)
[0.520s] 3: [parameter_bridge-3] [INFO] [1660866657.904733060] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/pose_transformstamped (ignition.msgs.Pose) -> /pose_transformstamped (geometry_msgs/msg/TransformStamped)] (Lazy 0)
[0.520s] 3: [parameter_bridge-3] [INFO] [1660866657.905200116] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/twist_twist (ignition.msgs.Twist) -> /twist_twist (geometry_msgs/msg/Twist)] (Lazy 0)
[0.521s] 3: [parameter_bridge-3] [INFO] [1660866657.905663295] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/vector3d_vector3 (ignition.msgs.Vector3d) -> /vector3d_vector3 (geometry_msgs/msg/Vector3)] (Lazy 0)
[0.521s] 3: [parameter_bridge-3] [INFO] [1660866657.906187911] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/wrench_wrench (ignition.msgs.Wrench) -> /wrench_wrench (geometry_msgs/msg/Wrench)] (Lazy 0)
[0.522s] 3: [parameter_bridge-3] [INFO] [1660866657.906668835] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/odometry_odometry (ignition.msgs.Odometry) -> /odometry_odometry (nav_msgs/msg/Odometry)] (Lazy 0)
[0.522s] 3: [parameter_bridge-3] [INFO] [1660866657.907486700] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/any_parametervalue (ignition.msgs.Any) -> /any_parametervalue (rcl_interfaces/msg/ParameterValue)] (Lazy 0)
[0.523s] 3: [parameter_bridge-3] [INFO] [1660866657.908092042] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/contact_contact (ignition.msgs.Contact) -> /contact_contact (ros_ign_interfaces/msg/Contact)] (Lazy 0)
[0.524s] 3: [parameter_bridge-3] [INFO] [1660866657.908955691] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/contacts_contacts (ignition.msgs.Contacts) -> /contacts_contacts (ros_ign_interfaces/msg/Contacts)] (Lazy 0)
[0.524s] 3: [parameter_bridge-3] [INFO] [1660866657.909584047] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/dataframe_dataframe (ignition.msgs.Dataframe) -> /dataframe_dataframe (ros_ign_interfaces/msg/Dataframe)] (Lazy 0)
[0.525s] 3: [parameter_bridge-3] [INFO] [1660866657.910147208] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/entity_entity (ignition.msgs.Entity) -> /entity_entity (ros_ign_interfaces/msg/Entity)] (Lazy 0)
[0.525s] 3: [parameter_bridge-3] [INFO] [1660866657.910673342] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/guicamera_guicamera (ignition.msgs.GUICamera) -> /guicamera_guicamera (ros_ign_interfaces/msg/GuiCamera)] (Lazy 0)
[0.526s] 3: [parameter_bridge-3] [INFO] [1660866657.911251488] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/jointwrench_jointwrench (ignition.msgs.JointWrench) -> /jointwrench_jointwrench (ros_ign_interfaces/msg/JointWrench)] (Lazy 0)
[0.527s] 3: [parameter_bridge-3] [INFO] [1660866657.911812095] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/light_light (ignition.msgs.Light) -> /light_light (ros_ign_interfaces/msg/Light)] (Lazy 0)
[0.527s] 3: [parameter_bridge-3] [INFO] [1660866657.912384571] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/param_paramvec (ignition.msgs.Param) -> /param_paramvec (ros_ign_interfaces/msg/ParamVec)] (Lazy 0)
[0.528s] 3: [parameter_bridge-3] [INFO] [1660866657.912920157] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/param_v_paramvec (ignition.msgs.Param_V) -> /param_v_paramvec (ros_ign_interfaces/msg/ParamVec)] (Lazy 0)
[0.528s] 3: [parameter_bridge-3] [INFO] [1660866657.913340861] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/stringmsg_v_stringvec (ignition.msgs.StringMsg_V) -> /stringmsg_v_stringvec (ros_ign_interfaces/msg/StringVec)] (Lazy 0)
[0.529s] 3: [parameter_bridge-3] [INFO] [1660866657.913857362] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/trackvisual_trackvisual (ignition.msgs.TrackVisual) -> /trackvisual_trackvisual (ros_ign_interfaces/msg/TrackVisual)] (Lazy 0)
[0.529s] 3: [parameter_bridge-3] [INFO] [1660866657.914387571] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/videorecord_videorecord (ignition.msgs.VideoRecord) -> /videorecord_videorecord (ros_ign_interfaces/msg/VideoRecord)] (Lazy 0)
[0.530s] 3: [parameter_bridge-3] [INFO] [1660866657.914914029] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/clock_clock (ignition.msgs.Clock) -> /clock_clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[0.530s] 3: [parameter_bridge-3] [INFO] [1660866657.915497495] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/batterystate_batterystate (ignition.msgs.BatteryState) -> /batterystate_batterystate (sensor_msgs/msg/BatteryState)] (Lazy 0)
[0.531s] 3: [parameter_bridge-3] [INFO] [1660866657.916391115] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/camerainfo_camerainfo (ignition.msgs.CameraInfo) -> /camerainfo_camerainfo (sensor_msgs/msg/CameraInfo)] (Lazy 0)
[0.532s] 3: [parameter_bridge-3] [INFO] [1660866657.916928916] [ros_ign_bridge]: Creating IGN->ROS Bridge: [/fluidpressure_fluidpressure (ignition.msgs.FluidPressure) -> /fluidpressure_fluidpressure (sensor_msgs/msg/FluidPressure)] (Lazy 0)
[... cut ...]
[20.436s] 3: ----------------------------------------------------------------------
[20.436s] 3: Ran 1 test in 0.000s
[20.436s] 3: 
[20.436s] 3: FAILED (failures=1)
[20.452s] 3: -- run_test.py: return code 1
[20.452s] 3: -- run_test.py: verify result file '/home/anderson/wec/igngzb/openrobotics/workspace/build/ros_ign_bridge/test_results/ros_ign_bridge/test_launch_test_ros_subscriber.launch.py.xunit.xml'
[20.458s]  3/13 Test  #3: test_launch_test_ros_subscriber.launch.py ...***Failed   20.29 sec
```

quarkytale, 2022-08-25:
I was not able to reproduce this, is the same test failing every time?

andermi, 2022-08-25:
> same test failing every time

It's the same tests within `test_launch_test_ros_subscriber.launch.py` that fail, but they don't fail every time.

quarkytale, 2022-08-26:
A suggestion was to check if its a flaky test or not by trying `--retest-until-pass N`.

mjcarroll, 2022-09-23:
@andermi out of curiosity, is there another bridge/publisher/subscriber instance that maybe a zombie?  I have had this happen rarely, but it seems to always be related to a test that didn't bring itself down cleanly.

mjcarroll, 2022-09-23:
Alright, I'm trying to prune down issues, so if you haven't seen this for a while, I'm going to close it.  If it comes up again, feel free to re-open.

closed 2022-09-23 by mjcarroll