gazebosim/ros_gz #290: Fail to create bridge for NavSatFix/NavSat messages

JasonRBowling, 2022-08-21:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
 Binary. Fortress version  6.11 with ROS2 Galactic, both installed via apt-get

## Description
* Expected behavior: I expected to be able to bridge navsat messages from Fortress to ROS2, as I can with other topics (odom, tf, clock. camera, lidar, IMU, empty all work as expected)

* Actual behavior:  Launching via launch file or command line gives:
"[parameter_bridge-2] Failed to create a bridge for topic [/r1/gps] with ROS2 type [sensor_msgs/msg/NavSatFix] and Ignition Transport type [ignition.msgs.NavSat]"

Example command line:
ros2 run ros_ign_bridge parameter_bridge /r1/gps@sensor_msgs/msg/NavSatFix@ignition.msgs.NavSat

Launch file snippet:

# Bridge
    bridge = Node(
        package='ros_ign_bridge',
        executable='parameter_bridge',
        arguments=['/clock@rosgraph_msgs/msg/Clock[ignition.msgs.Clock',
                   '/model/robot1/tf@tf2_msgs/msg/TFMessage[ignition.msgs.Pose_V',
                   '/model/robot2/tf@tf2_msgs/msg/TFMessage[ignition.msgs.Pose_V',
                   '/model/robot1/cmd_vel@geometry_msgs/msg/Twist@ignition.msgs.Twist',
                   '/model/robot1/odometry@nav_msgs/msg/Odometry@ignition.msgs.Odometry',
                   '/model/robot2/cmd_vel@geometry_msgs/msg/Twist@ignition.msgs.Twist',
                   '/model/robot2/odometry@nav_msgs/msg/Odometry@ignition.msgs.Odometry',
                   '/world/diff_drive/model/robot1/link/base_link/sensor/imu_sensor/imu@sensor_msgs/msg/Imu@ignition.msgs.IMU',
                   '/world/diff_drive/model/robot2/link/base_link/sensor/imu_sensor/imu@sensor_msgs/msg/Imu@ignition.msgs.IMU',
                   '/world/diff_drive/model/robot1/link/base_link/sensor/camera_front/image@sensor_msgs/msg/Image@ignition.msgs.Image',
                   '/world/diff_drive/model/robot2/link/base_link/sensor/camera_front/image@sensor_msgs/msg/Image@ignition.msgs.Image',
                   '/world/diff_drive/model/robot1/link/base_link/sensor/base_scan/scan@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan',
                   '/world/diff_drive/model/robot2/link/base_link/sensor/base_scan/scan@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan',
                   '/r1/deploy@std_msgs/msg/Empty@ignition.msgs.Empty',
                   '/r2/deploy@std_msgs/msg/Empty@ignition.msgs.Empty',
                   '/r1/gps@sensor_msgs/msg/NavSatFix[ignition.msgs.NavSat'],

        output='screen',
        remappings=[('/model/robot1/cmd_vel', '/robot1/cmd_vel'),
                    ('/model/robot2/cmd_vel', '/robot2/cmd_vel'),
                    ('/model/robot1/tf', '/robot1/tf'),
                    ('/model/robot2/tf', '/robot2/tf'),
                    ('/model/robot1/odometry', '/robot1/odom'),
                    ('/model/robot2/odometry', '/robot2/odom'),
                    ('/world/diff_drive/model/robot1/link/base_link/sensor/imu_sensor/imu', '/robot1/imu'),
                    ('/world/diff_drive/model/robot2/link/base_link/sensor/imu_sensor/imu', '/robot2/imu'),
                    ('/world/diff_drive/model/robot1/link/base_link/sensor/base_scan/scan', '/robot1/scan'),
                    ('/world/diff_drive/model/robot2/link/base_link/sensor/base_scan/scan', '/robot2/scan'),
                    ('/world/diff_drive/model/robot1/link/base_link/sensor/camera_front/image', '/robot1/camera_front/image'),
                    ('/world/diff_drive/model/robot2/link/base_link/sensor/camera_front/image', '/robot2/camera_front/image')
                    ]

    )

## Steps to reproduce

1. Define navsat sensor on robot in Fortress - I confirmed that the ign/gazebo topic was working with "ign topic -e -t /r1/gps"
2. Launch bridge: ros2 run ros_ign_bridge parameter_bridge /r1/gps@sensor_msgs/msg/NavSatFix@ignition.msgs.NavSat

Thank you!

chapulina, 2022-08-27:
> Binary. Fortress version 6.11 with ROS2 Galactic, both installed via apt-get

NavSat support was added to Galactic in https://github.com/gazebosim/ros_gz/pull/268, and we haven't made a release since. You could try building `ros_ign` from source.

Also note that `ros-galactic-ros-ign` uses Edifice, not Fortress, per [REP-2000](https://ros.org/reps/rep-2000.html#galactic-geochelone-may-2021-november-2022).

closed 2022-08-27 by chapulina

JasonRBowling, 2022-08-27:
I see. Thank you!