gazebosim/ros_gz #291: Bring back -J option to specify joint angles when creating model

rcywongaa, 2022-09-07:
## Desired behavior
In Gazebo Classic + ROS1, there was the ability to use [`spawn_model`](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_ros/scripts/spawn_model#L78) with `-J` to specify the joint values of the spawned model, albeit some [never resolved bugs](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/93)...

This seems to be missing in the current Ignition Gazebo + ROS2 method of spawning models via [`create`](https://github.com/gazebosim/ros_gz/blob/melodic/ros_ign_gazebo/src/create.cpp).  Given how useful it is to spawn a model at a specific pose (e.g. if the null pose results in collision), it would be nice to have that feature back (and ideally with that bug fixed :smiley: )

mikramarc, 2023-12-14:
HI [rcywongaa](https://github.com/rcywongaa) any idea about the status of that? Or have you maybe found a reasonable workaround?

azeey, 2023-12-20:
> HI [rcywongaa](https://github.com/rcywongaa) any idea about the status of that? Or have you maybe found a reasonable workaround?

If you're using the `JointTrajectoryController` system, one possible workaround is to put the initial joint positions in the `<initial_position>` parameter of the system. See https://gazebosim.org/api/sim/7/classgz_1_1sim_1_1systems_1_1JointTrajectoryController.html#:~:text=initial_position. The `JointPositionController` also has that parameter.

g1y5x3, 2025-02-08:
If I would like to use `ApplyJointForce` instead, is there still a way for me to set `<initial_position>`? Was trying to implement a MPC controller that outputs torque