gazebosim/ros_gz #300: Release for rolling?

ijnek, 2022-09-20:
In a project I have, I was using ros_ign_bridge as a library (ie. ``#include "ros_ign_bridge/convert/sensor_msgs.hpp``), rather than as a standalone node. 

Unfortunately, this broke on humble during the transition from ros_ign_bridge -> ros_gz_bridge. So I decided to change all ros_ign_bridge -> ros_gz_bridge in my project, but then now the rolling CI (which ran on the same branch) is failing because a release hasn't been made since the rename.

I could go about splitting up my humble and rolling branch and running CI separately, but the problem will disappear once ros_gz_bridge is available in rolling.

Could I request the maintainers for a release for rolling please?

mjcarroll, 2022-09-23:
I have a large port from galactic to rolling to get in (#275) and then I think we will be good for a release.

ijnek, 2022-12-09:
Hi @mjcarroll, do you have any updates on making a release?

For a while, it's been like this:
* galactic / **rolling**: ros_ign
* humble: ros_gz (or ros_ign with a deprecation warning)

A release would help cutting-edge rolling users a lot. What are the roadblockers?

mjcarroll, 2022-12-09:
I think we can manage a release.  @ahcorde any reason not to?

ijnek, 2022-12-22:
Ping @ahcorde, thanks

christophfroehlich, 2023-04-17:
Is this correct that it is still not released for rolling?

Furthermore, in contrast to the compatibility table in the README.md the ros2 branch is not compatible with fortress any more -> I have to use humble branch to compile it for rolling+fortress. Or have I missed something?

https://github.com/gazebosim/ros_gz/blob/7c62a1ce3cfb4eb3b54e1cfd092c9b63a9d5f255/ros_gz_sim/CMakeLists.txt#L38

mjcarroll, 2023-05-05:
For humble, you should now be using the `humble` branch.  I am making a release for rolling/iron and fortress.

mjcarroll, 2023-05-05:
Fixing branch names in #387 and #388

azeey, 2023-05-08:
I think it's okay to do Fortress/Rolling binary releases while Iron is still pending, but once it's released, we should switch to preparing for the Gazebo Harmonic/ROS J-turtle pairing. So in that sense, the `This branch is only compatible with Gazebo Garden forward.` message seems correct, albeit a little early.

moriarty, 2023-05-08:
> I think it's okay to do Fortress/Rolling binary releases while Iron is still pending, but once it's released, we should switch to preparing for the Gazebo Harmonic/ROS J-turtle pairing. So in that [...]

:+1: If the `Fortress/Rolling` binary release also has a `rolling` branch that source builds can depend on similar to the `humble` branch then that sounds great

closed 2024-04-21 by ijnek