gazebosim/ros_gz #307: ros_gz_bridge: config file broken with Humble release

oKermorgant, 2022-10-04:
## Environment
* OS Version: 22.04
* ROS2 : Humble
* Gazebo : Fortress (binary) + quite recent `ros-humble-ros-gz-bridge`

## Description

I am using the `config_file` approach for `ros_gz_bridge`, as some GZ topics have non-valid ROS names (e.g. [JointPositionController](https://gazebosim.org/api/gazebo/6.1/classignition_1_1gazebo_1_1systems_1_1JointPositionController.html) that includes a `/0/` in the name, which is non acceptable even if it is remapped)

When moving from galactic to humble, I have relied on the source version of `ros_gz` and simply adapting the syntax was fine: https://github.com/gazebosim/ros_gz/pull/303

A few days ago the `ros-humble-ros-gz` packages were released, but it seems that the `config_file` approach does not work with the binary. Using the current repo with the `humble` branch still works fine.

* Expected behavior: bridge should forward messages between Gazebo and ROS 2
* Actual behavior: bridge works with command-line configuration but not with `config_file` parameter

## Steps to reproduce
1. Run a basic Gazebo simulation
2. Run e.g. `ros2 run ros_gz_bridge parameter_bridge /world/<world>/clock@rosgraph_msgs/msg/Clock@ignition.msgs.Clock` (the `/clock` topic seems to be world-prefixed now?): the same topic appears on ROS 2
3. Run e.g. `ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=<file given below>` : the `/clock` topic does not appear on ROS 2
```
- gz_topic_name: /world/<world>/clock
  gz_type_name: ignition.msgs.Clock
  ros_topic_name: /clock  # remapping
  ros_type_name: rosgraph_msgs/msg/Clock
  direction: GZ_TO_ROS
```

## Alternative

It would be perfectly fine using only the command line configuration, but some plugins (JointPositionController) use invalid ROS topics. A way to pick the topic for such plugins would be an acceptable workaround.

mjcarroll, 2022-10-05:
This is likely something I introduced in a forward-port, I will look into it.

mjcarroll, 2022-10-11:
> the /clock topic seems to be world-prefixed now?

gz-sim will make `/clock` a root topic as long as no other publisher is detected on `/clock`

mjcarroll, 2022-10-11:
I'm having a bit of an issue trying to reproduce this with the humble branch:

Command line:
```
$ ros2 run ros_gz_bridge parameter_bridge /clock@rosgraph_msgs/msg/Clock@ignition.msgs.Clock 
[INFO] [1665499487.952566029] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
[INFO] [1665499487.953077017] [ros_gz_bridge]: Creating ROS->GZ Bridge: [/clock (rosgraph_msgs/msg/Clock) -> /clock (ignition.msgs.Clock)] (Lazy 0)

$ ros2 topic info /clock
Type: rosgraph_msgs/msg/Clock
Publisher count: 1
Subscription count: 1
```

Config file:
```
$ cat test.config 
- gz_topic_name: /clock
  gz_type_name: ignition.msgs.Clock
  ros_topic_name: /clock  # remapping
  ros_type_name: rosgraph_msgs/msg/Clock
  direction: GZ_TO_ROS

$ ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=test.config
[INFO] [1665499570.382543769] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)

$ ros2 topic info /clock
Type: rosgraph_msgs/msg/Clock
Publisher count: 1
Subscription count: 0
  ```

mjcarroll, 2022-10-11:
@oKermorgant if it is possible that you were still using `ign_type_name` and `ign_topic_name` in your configuration files, I have addressed that with deprecation warnings in #310

mjcarroll, 2022-10-11:
Ha! Actually we need a humble release, it seems.

oKermorgant, 2022-10-11:
@mjcarroll  thanks for your time.
I can still reproduce the issue when relying on the `ros-humble-gz-bridge` binary, while the behavior is as expected when using the package compiled from source.

```
~/code/libs/ros2$ ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=./test.yaml # running source-compiled ros_gz_bridge
[INFO] [1665504213.150291024] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/clock (ignition.msgs.Clock) -> /clock (rosgraph_msgs/msg/Clock)] (Lazy 0)
^C[INFO] [1665504214.763871376] [rclcpp]: signal_handler(signum=2)
~/code/libs/ros2$ mv install/ros_gz_bridge/ .   # removing ros_gz_bridge from my overlay
~/code/libs/ros2$ ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=./test.yaml # running from /opt/ros/humble
# no [INFO] and no bridge
^C[INFO] [1665504239.133323306] [rclcpp]: signal_handler(signum=2)
```
I have indeed used  `gz_*` keys in the configuration file, a deprecation warning will be welcome anyway!
I cannot see how it could be specific to my computer, but maybe it is.

mjcarroll, 2022-10-11:
> I can still reproduce the issue when relying on the ros-humble-gz-bridge binary, while the behavior is as expected when using the package compiled from source.

Yep, that was my mistake, I believe the configuration bits are completely excluded from the previous humble release.  Once #309 and #310 get in, we can make a new release which will include the configuration as well as those fixes.

mjcarroll, 2022-10-12:
Humble release will address this: https://github.com/ros/rosdistro/pull/34912

mjcarroll, 2022-10-14:
rosdistro pr merged, it will go out in the next Humble sync, so I'm going to close this.  Thanks for iterating with me.

closed 2022-10-14 by mjcarroll

oKermorgant, 2022-10-17:
Cool, thanks.