gazebosim/ros_gz #308: ros_ign_gazebo- `ign_args` launch arg not respected in `ign_gazebo.launch.py`

mwcondino, 2022-10-07:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
`ros-humble-ros-ign-gazebo` at `0.244.6-1jammy.20220916.172640`
`libignition-gazebo6:amd64` at `6.12.0-1~jammy`

## Description
It looks like the `ign_gazebo.launch.py` launch file may not respect `ign_args` in a humble environment with 

## Steps to reproduce

1. Ensure `ros-humble-ros-ign-gazebo` is installed
2. Try running a gazebo session headless through the launch file: 
```
ros2 launch ros_ign_gazebo ign_gazebo.launch.py ign_args:="-s"
```
3. You should see a gui pop up after this - this indicates the `ign_args` parameter is not set.

## Workaround
Using `gz_args` instead of `ign_args` appears to be working. In addition, I think if you force the ignition version like this....
```
ros2 launch ros_ign_gazebo ign_gazebo.launch.py ign_args:="-s" ign_version:=6
```
the bug is no longer evident.

It's probably best to update launch files to use `gz_args` as a fix here, since that is more future facing and `ign_*` args are deprecated.

mjcarroll, 2022-10-11:
@mwcondino please look at #309 and see if this resolves the issue.

mjcarroll, 2022-10-11:
> It's probably best to update launch files to use gz_args as a fix here, since that is more future facing and ign_* args are deprecated.

This is ideal long-term, but there should be some middle ground with deprecation warnings, that is what I have implemented.

mjcarroll, 2022-10-14:
#309 has been released to humble and will be available in the next sync.  If this doesn't resolve it, please feel free to reopen.

closed 2022-10-14 by mjcarroll

mwcondino, 2022-10-15:
Thanks @mjcarroll, that fix looks good to me!

PR #309 (2022-10-12): Fix launch substitutions for ign_args
# 🦟 Bug fix

Fixes #308 

## Summary

Improves the launch file to correctly check for `ign_args` in addition to `gz_args`.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While w [... cut]