gazebosim/ros_gz #318: Diff Drive demo not working with Gazebo Garden

Myzhar, 2022-10-28:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
Source, branch `ros2`

## Description
* Expected behavior: The `diff_drive.launch.py` demo correctly works in Gazebo and Rviz2
* Actual behavior:  The communication between Gazebo and ROS2 is not established

## Steps to reproduce

1. Install Gazebo Garden
2. clone the `ros_gz` repository, branch `ros2`
3. build
4. launch the diff drive demo: `$ ros2 launch ros_gz_sim_demos diff_drive.launch.py`

## Output
```
[parameter_bridge-2] [INFO] [1666959084.759484057] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle_blue/cmd_vel (ignition.msgs.Twist) -> /model/vehicle_blue/cmd_vel (geometry_msgs/msg/Twist)] (Lazy 0)
[parameter_bridge-2] [WARN] [1666959084.759605733] [ros_gz_bridge]: Failed to create a bridge for topic [/model/vehicle_blue/cmd_vel] with ROS2 type [/model/vehicle_blue/cmd_vel] to topic [geometry_msgs/msg/Twist] with Gazebo Transport type [ignition.msgs.Twist]
[parameter_bridge-2] [INFO] [1666959084.759618904] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle_blue/odometry (ignition.msgs.Odometry) -> /model/vehicle_blue/odometry (nav_msgs/msg/Odometry)] (Lazy 0)
[parameter_bridge-2] [WARN] [1666959084.759635260] [ros_gz_bridge]: Failed to create a bridge for topic [/model/vehicle_blue/odometry] with ROS2 type [/model/vehicle_blue/odometry] to topic [nav_msgs/msg/Odometry] with Gazebo Transport type [ignition.msgs.Odometry]
[parameter_bridge-2] [INFO] [1666959084.759642196] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle_green/cmd_vel (ignition.msgs.Twist) -> /model/vehicle_green/cmd_vel (geometry_msgs/msg/Twist)] (Lazy 0)
[parameter_bridge-2] [WARN] [1666959084.759652507] [ros_gz_bridge]: Failed to create a bridge for topic [/model/vehicle_green/cmd_vel] with ROS2 type [/model/vehicle_green/cmd_vel] to topic [geometry_msgs/msg/Twist] with Gazebo Transport type [ignition.msgs.Twist]
[parameter_bridge-2] [INFO] [1666959084.759658763] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/model/vehicle_green/odometry (ignition.msgs.Odometry) -> /model/vehicle_green/odometry (nav_msgs/msg/Odometry)] (Lazy 0)
[parameter_bridge-2] [WARN] [1666959084.759668489] [ros_gz_bridge]: Failed to create a bridge for topic [/model/vehicle_green/odometry] with ROS2 type [/model/vehicle_green/odometry] to topic [nav_msgs/msg/Odometry] with Gazebo Transport type [ignition.msgs.Odometry]
```

## Solution 
in file `diff_drive.launch.py` replace
```
    # Bridge
    bridge = Node(
        package='ros_gz_bridge',
        executable='parameter_bridge',
        arguments=['/model/vehicle_blue/cmd_vel@geometry_msgs/msg/Twist@ignition.msgs.Twist',
                   '/model/vehicle_blue/odometry@nav_msgs/msg/Odometry@ignition.msgs.Odometry',
                   '/model/vehicle_green/cmd_vel@geometry_msgs/msg/Twist@ignition.msgs.Twist',
                   '/model/vehicle_green/odometry@nav_msgs/msg/Odometry@ignition.msgs.Odometry'],
        parameters=[{'qos_overrides./model/vehicle_blue.subscriber.reliability': 'reliable',
                     'qos_overrides./model/vehicle_green.subscriber.reliability': 'reliable'}],
        output='screen'
    )
```
with
```
    # Bridge
    bridge = Node(
        package='ros_gz_bridge',
        executable='parameter_bridge',
        arguments=['/model/vehicle_blue/cmd_vel@geometry_msgs/msg/Twist@gz.msgs.Twist',
                   '/model/vehicle_blue/odometry@nav_msgs/msg/Odometry@gz.msgs.Odometry',
                   '/model/vehicle_green/cmd_vel@geometry_msgs/msg/Twist@gz.msgs.Twist',
                   '/model/vehicle_green/odometry@nav_msgs/msg/Odometry@gz.msgs.Odometry'],
        parameters=[{'qos_overrides./model/vehicle_blue.subscriber.reliability': 'reliable',
                     'qos_overrides./model/vehicle_green.subscriber.reliability': 'reliable'}],
        output='screen'
    )
```

Myzhar, 2022-10-28:
Additional note: the same issue is present in all the other demos:
* air_pressure.launch.py
* battery.launch.py
* camera.launch.py
* depth_camera.launch.py
* gpu_lidar_bridge.launch.py
* gpu_lidar.launch.py
* imu.launch.py
* joint_states.launch.py
* joint_states.launch.py
* navsat.launch.py
* rgbd_camera_bridge.launch.py
* rgbd_camera.launch.py
* sdf_parser.launch.py
* tf_bridge.launch.py
* triggered_camera.launch.py

ahcorde, 2022-10-28:
@Myzhar, are you planning to open a PR? otherwise I will add the changes.

Myzhar, 2022-10-28:
No @ahcorde I'm not going to open a PR now.
I'm struggling with my simulator plug-in code and I'm afraid I don't spend enough time on the PR.

closed 2022-10-28 by ahcorde