gazebosim/ros_gz #325: GPU Lidar sensor bridge is not passing all of the sensor data from the GZ topic to the defined ROS2 topic.

ConnorDTaylor, 2022-11-02:
## Environment
* OS Version: Ubuntu 22.04
ROS2: Humble, binary install
Gazebo: Gazebo Garden, binary install
ros_ign branch: ros2 (master branch, re-cloned today)
## Description
* Expected behavior: When creating the bridge between the GPU_Lidar sensor and Gazebo Garden, I would expect the horizontal and vertical samples from the sesnor to be passed to the defined Ros2 topic.
* Actual behavior: When I try to pass the sensor measurements to Ros2 the only lidar measurements being received are from the first vertical sample and that array of horizontal samples. For example, if I define the GPU Lidar to have 10 horizontal and 10 vertical samples, it generates 100 total Lidar points. However, I am only getting 10 of them passed to ROS. 

## Steps to reproduce
1. Make a world with the GPU lidar sensor
2. In the launch file for the world, use the bridge to send the gazebo topic data to ROS.
3. With some parameter changes it will be noticed that increasing the number of vertical samples increases the number of ranges in the gz topic, but not the number of ranges in the ROS2 topic. 

## Output
Here I will prove images of the SDF file for my sensor, the bridge in my launch file, and the GZ topic console output vs the ROS2 topic console output.
[image]
[image]
[image]
[image]

mjcarroll, 2022-11-02:
Yes please, if you can open a pull request, I would be happy to review it.

mjcarroll, 2023-04-17:
I replied here: https://github.com/gazebosim/ros_gz/pull/326#issuecomment-1511623123

But in summary, a single "row" of lidar data is expected for a LaserScan message.  If you want the vertical and horizontal ranges, then a PointCloud message is what you are looking for.

closed 2023-04-17 by mjcarroll