gazebosim/ros_gz #338: The ros2 branch is compatible with Gazebo Garden forward but README says otherwise

bjsowa, 2022-12-19:
According the README the `ros2` branch should work with Gazebo Edifice and Fortress but this line says otherwise:
https://github.com/gazebosim/ros_gz/blob/28b0eb66741b770e18e3aba1a8cf692f7b609794/ros_gz_sim/CMakeLists.txt#L38

Also, the current development branch for ROS2 Humble is set to `humble` in rosdistro but in the README it is specified as `ros2`.

Could you please make it clear which branch should be used with what Distros and what ROS Distro - Gazebo version combinations are supported?

Ryanf55, 2023-01-31:
Is this going to solve the fact that the README for `humble` says `Garden` is not released, the [docs](https://gazebosim.org/docs/garden) say it is upcoming release, but OSRF says it was released [here](https://www.openrobotics.org/blog/2022/10/3/gazebo-garden-released)?

azeey, 2024-01-04:
The `ros2` branch should now be up-to-date. There's also a branch for `humble` and `iron` now, so it should hopefully be less confusing.

closed 2024-01-04 by azeey