gazebosim/ros_gz #340: ros noetic support for gazebo-garden

julled, 2022-12-22:
## Desired behavior
Quite some users still depend on ROS1 environments. The latest gazebo garden version is not supported for noetic yet.

## Alternatives considered
Switching to ROS2 is not an option for some people.

## Implementation suggestion

## Additional context

mjcarroll, 2023-01-10:
There isn't a technical reason that this cannot be supported, it has just been largely outside of the scope of our current project workload.

The simplest approach to trying the garden/noetic combination would be to copy the logic in the CMakeLists.txt file to also accommodate garden versions.

The logic in question is here:

https://github.com/gazebosim/ros_gz/blob/349b36daa01ee21b7d1de265c8015e47230f6c97/ros_ign_bridge/CMakeLists.txt#L23-L32

Note that you will likely get deprecation warnings as part of the build/run as the `noetic` branch of the bridge hasn't been adjusted for the renaming that has occurred between gazebo fortress and garden.

For now, this will likely remain on our maintainers' backlog, but if you are interested in picking it up, I would be happy to assign it to you.

Tuxliri, 2024-01-07:
I have started doing some work on this, PR #484