gazebosim/ros_gz #344: configure ros_gz_bridge to modify frame_ids

0x4d494b, 2023-01-04:
I am inserting a robot model into Gazebo 7  using robot_state_publisher. Then, I am bridging sensors back to ROS2. The sensor IDs have a compound frame_id name (model_name/base_frame/sensor_name) that don't match link names in URDF. That causes all kinds of issues if I try launching RVIZ with the simulation. 

This also applies to pose publisher plugin in Gazebo, so getting transformations for everything that I need by modifying link names, for example (sensors, poses, etc)  so it would match the robot_state_publisher model becomes tricky.

## Desired behavior
ros_gz_bridge could have a config parameter to strip the names up to last "::" instead of replacing those with a '/'. Thoughts?

azeey, 2023-01-06:
I believe you can set the frame_id of the sensor to whatever you want with `gz_frame_id` (see https://github.com/gazebosim/gz-sensors/pull/195)

0x4d494b, 2023-01-17:
> I believe you can set the frame_id of the sensor to whatever you want with `gz_frame_id` (see [gazebosim/gz-sensors#195](https://github.com/gazebosim/gz-sensors/pull/195))

Thank you very much for the info. It works for sensors and is exactly what I needed. However, the pose publisher doesn't seem to have the same feature.

samprt-heracles, 2023-02-10:
This doesn't seem to be working with Gazebo Garden and ROS2 Humble. As mentionned in [gazebosim/gz-sensors#195](https://github.com/gazebosim/gz-sensors/pull/195), Gazebo throws a warning and the published frame_id is still the original one.
I may be doing something wrong though.

0x4d494b, 2023-02-11:
> This doesn't seem to be working with Gazebo Garden and ROS2 Humble. 
If you use packages from ubuntu then they don't include the latest changes, but if you use ign_frame_id it should work