gazebosim/ros_gz #346: CameraInfo not published to ROS Noetic unless the image is viewed in the Gazebo GUI

alex-ssom, 2023-01-06:
## Environment
* OS Version: Ubuntu 20.04.5 LTS x86_64
* Source or binary build? Gazebo Fortress from source (as of yesterday) 

## Description
* Expected behavior: The CameraInfo should be published as soon as the ros-ign-bridge has registered a ROS listener for the image topic.
* Actual behavior: You need to manually view the image in the Gazebo GUI to trigger the publishing of the CameraInfo

## Steps to reproduce
1. Open a Gazebo world with a camera
2. Run the ros-ign-bridge to publish the CameraInfo topic
3. `rostopic echo` the CameraInfo topic
4. BUG: No message is published
5. In the Gazebo GUI, add `Image Display`
6. Select the camera and hit the refresh button
7. The CameraInfo should now be published

Note: This might not be completely related to the ros_gz_bridge. Since this issue also happens with the following steps:
1. Open a Gazebo world with a camera
2. `ign topic -l`
3. `ign topic -e -n 1 -t camera/camera_info`
4. No data is echoed
5. Click on the refresh button in `Image Display`
6. The data should now be echoed in the terminal

## Output

Before refresh:
[image]

After:
[image]

iche033, 2023-01-06:
likely an issue in gz-sensors. Camera sensors do not generate / publish any data if there are no subscribers to the image topic. The [CameraSensor::HasConnections](https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/src/CameraSensor.cc#L823) only checks for subscribers to image data but not camera info. One potential fix is to also check camera info topic subscribers and in `CameraSensor::HasConnections` and change `CameraSensor::Update` function to publish only camera info data (and skip rendering image data) if there are only camera info subscribers.

alex-ssom, 2023-01-09:
Should I open a new issue in gz-sensors?

azeey, 2023-01-09:
Closing because the root issue is in gz-sensors

closed 2023-01-09 by azeey