gazebosim/ros_gz #354: ros_gz_bridge parameter_bridge not parsing yaml files to create parameter bridges.

Jack17432, 2023-01-23:
## Environment
* OS Version: Ubuntu 22.04
* ROS2 version: humble
* Source: 6c2e56efbaadf6e3683d7817703b1b626b768726, ros2

## Description
* Expected behavior: ros_gz_bridge parses YAML file to load up a list of bridges
* Actual behavior: [readFromYaml]: Could not parse config, top level must be a YAML sequence

## Steps to reproduce

1. download **any** of the yaml examples in the ros_gz_bridge example files
2. Use parameter_bridge from ros_gz_bridge use pass in 'config_file': path_to_config
3. Run ros2 launch file

## Output
[image]
[image]

mjcarroll, 2023-01-23:
Thanks for reporting, taking a look

mjcarroll, 2023-01-23:
What happens if you run from command line, for example:

```
ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=./src/ros_gz/ros_gz_bridge/test/config/full.yaml
```

Jack17432, 2023-01-23:
[image]

mjcarroll, 2023-01-24:
Okay, this indicates that it can find this file when loaded from the command line.  I believe the error you are seeing is because the configuration is pointing to a file that does not exist.

I'm going to open a PR to make the error condition more explicit when you are trying to open a non-existent file.

The failure to load with `ignition.` prefixed messages is a second fix that I will need to add to maintain compatibility with the `humble` branch.

mjcarroll, 2023-03-02:
@Jack17432 friendly ping, can you confirm that you were able to get this working with the improved error messages?

TanguyGerniers, 2023-06-24:
Hi @mjcarroll, I've encountered the same error trying to run a yaml file : 

```
~$ ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=~/gazebo_ws/ros_bridge/config.yaml
[ERROR] [1687567401.137092983] [readFromYaml]: Could not parse config, top level must be a YAML sequence
[ERROR] [1687567402.136611339] [readFromYaml]: Could not parse config, top level must be a YAML sequence
```
I added the 'improved error messages" changes you made, to no effect.

the contents of config.yaml are just : 

```
- topic_name: "/clock"
  ros_type_name: "rosgraph_msgs/msg/Clock"
  gz_type_name: "gz..msgs.Clock"
  subscriber_queue: 10       # Default 10
  publisher_queue: 10        # Default 10
  lazy: false                # Default "false"
  direction: GZ_TO_ROS       # Default "BIDIRECTIONAL" - Bridge both directions
                             # "GZ_TO_ROS" - Bridge GZ topic to ROS
                             # "ROS_TO_GZ" - Bridge ROS topic to GZ
```
Am I supposed to place that yaml file somewhere in particular?

azeey, 2023-06-24:
Doesn't the `GZ_TO_ROS` need to be in quotes?

TanguyGerniers, 2023-06-24:
> Hi @azeey, thank you for your reply. GZ_TO_ROS isnt in quotes in Jack's .yaml further up nor in the [example code for the ros_gz_bridge](https://github.com/gazebosim/ros_gz/blob/humble/ros_gz_bridge/README.md) and the [test configs](https://github.com/gazebosim/ros_gz/blob/ros2/ros_gz_bridge/test/config/full.yaml), but I have tried doing so just in case, with the same outcome.

mjcarroll, 2023-10-31:
Correct, "GZ_TO_ROS" should be in quotes.

closed 2023-10-31 by mjcarroll

Ali-Hossam, 2024-06-14:
If anyone happens to encounter the same error, make sure that you added the config folder that has the yaml file to CMakeLists install Directory.

MnM03, 2024-06-30:
Did someone solved this issue? I am having the same problem.

anushkav1888, 2024-07-18:
I was running parameter bridge node using a YAML config file through a python launch file, and encountered the same error.
Make sure your config exists in install directory inside your workspace, and you define config file path correctly in your launch file (specify using get_package_share_directory or any other equivalents) .

simona-99, 2024-09-24:
Hi,
I'm also getting the same error: ` [readFromYaml]: Could not parse config, top level must be a YAML sequence`
And when I run this: 
```
ros2 run ros_gz_bridge parameter_bridge --ros-args -p config_file:=./src/ros_gz/ros_gz_bridge/test/config/full.yaml
[ERROR] [1727164413.215385465] [readFromYaml]: Could not parse config, top level must be a YAML sequence
```
I made sure to include the config folder containing the .yaml file in the CMakeLists.txt file:
```
install(DIRECTORY config/
  DESTINATION share/${PROJECT_NAME}/config
)
```
I also made sure that the config file exists in the install directory inside my workspace. 

Thank you in advance for any help you can give me ;)

Kmedrano101, 2025-02-12:
As @Ali-Hossam and @anushkav1888  mentioned, solution is to make sure that the folder where your yaml file is be included into CMakeLists.txt : install(DIRECTORY config DESTINATION share/${PROJECT_NAME}) foy pyhon setup: data_files=[(os.path.join('share', package_name, 'resource'), glob('config/*.yaml'))]. Tested in Ubuntu 22.04, ROS2 humble.