gazebosim/ros_gz #358: Include errors compiling ros_gz_bridge on Humble

Ryanf55, 2023-02-02:
## Environment
* OS Version: Ubuntu 22.04
* ROS version: humble, binary install
* Source or binary build? -> Gazebo Binary 7.2.0 install

  

## Description
* Expected behavior: I can compile humble without include errros.
* Actual behavior:  ros_gz_bridge fails to compile due include errors in `gz/utilities/SuppressWarning.hh`

## Steps to reproduce

1. `mkdir -p gazebo_ws/src`
2. `cd gazebo_ws/src`
3. `git clone git clone https://github.com/gazebosim/ros_gz.git`
4. `cd ros_gz`
5. `git checkout humble`
6.  `source /opt/ros/humble/setup.bash`
7. `rosdep install -r --from-paths src -i -y` => Note, I omitted rolling since i'm using humble
8. `colcon build --packages-up-to ros_gz_bridge`

## Output

```
ubuntu@ubuntu22:~/Documents/garden_ws$ colcon build --packages-up-to ros_gz_sim_demos && . install/setup.bash
Starting >>> ros_gz_interfaces
Starting >>> ros_gz_sim
Finished <<< ros_gz_sim [0.38s]                                                                      
Finished <<< ros_gz_interfaces [1.54s]                     
Starting >>> ros_gz_bridge
[Processing: ros_gz_bridge]                             
[Processing: ros_gz_bridge]                                     
[Processing: ros_gz_bridge]                                       
[Processing: ros_gz_bridge]                                       
--- stderr: ros_gz_bridge                                       
In file included from /usr/include/ignition/transport11/gz/transport.hh:25,
                 from /usr/include/ignition/transport11/ignition/transport.hh:18,
                 from /home/ubuntu/Documents/garden_ws/build/ros_gz_bridge/generated/test/gz_subscriber.cpp:17:
/usr/include/ignition/transport11/gz/transport/Clock.hh:24:10: fatal error: gz/utilities/SuppressWarning.hh: No such file or directory
   24 | #include <gz/utilities/SuppressWarning.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/test_gz_subscriber.dir/build.make:122: CMakeFiles/test_gz_subscriber.dir/generated/test/gz_subscriber.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:449: CMakeFiles/test_gz_subscriber.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/ignition/transport11/gz/transport.hh:25,
                 from /usr/include/ignition/transport11/ignition/transport.hh:18,
                from /home/ubuntu/Documents/garden_ws/src/ros_gz/ros_gz_bridge/test/serverclient/gz_server.cpp:24:
/usr/include/ignition/transport11/gz/transport/Clock.hh:24:10: fatal error: gz/utilities/SuppressWarning.hh: No such file or directory
   24 | #include <gz/utilities/SuppressWarning.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/test_gz_server.dir/build.make:76: CMakeFiles/test_gz_server.dir/test/serverclient/gz_server.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:476: CMakeFiles/test_gz_server.dir/all] Error 2
In file included from /usr/include/ignition/transport11/gz/transport.hh:25,
                 from /usr/include/ignition/transport11/ignition/transport.hh:18,
                 from /home/ubuntu/Documents/garden_ws/build/ros_gz_bridge/generated/test/gz_publisher.cpp:18:
/usr/include/ignition/transport11/gz/transport/Clock.hh:24:10: fatal error: gz/utilities/SuppressWarning.hh: No such file or directory
   24 | #include <gz/utilities/SuppressWarning.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/test_gz_publisher.dir/build.make:122: CMakeFiles/test_gz_publisher.dir/generated/test/gz_publisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:395: CMakeFiles/test_gz_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ros_gz_bridge [2min 6s, exited with code 2]
Summary: 2 packages finished [2min 8s]
  1 package failed: ros_gz_bridge
  1 package had stderr output: ros_gz_bridge
  2 packages not processed
```

It appears that ignition is not exporting the include directories properly. Since I thought ignition name was retired, I'm a bit confused to see this.

azeey, 2023-02-03:
I built the code following your instructions without any issue in a docker container. My guess is you have stale ignition packages that needs to be updated. Can you check with `apt list --upgradable "libignition-*"`?

Also, note that you were building `ros_gz` for Fortress. If you're wanting to build it for Garden, you'll need to set `GZ_VERSION=garden".

Ryanf55, 2023-02-04:
Ok thanks, sorry, missed that export in the README. Trying again, here's the result: 
```
ryan@ryan-m93p:~/Documents/gz_ws$ apt list --upgradable "libignition-*"
Listing... Done
```

Note: The rosdep key for gz-transport12 and gz-math7 cannot be resolved. 
```
ryan@ryan-m93p:~/Documents/gz_ws$ rosdep install -r --from-paths src -i -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_gz_bridge: Cannot locate rosdep definition for [gz-transport12]
ros_gz_image: Cannot locate rosdep definition for [gz-transport12]
ros_gz_sim: Cannot locate rosdep definition for [gz-math7]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
Even after I follow the [documentation](https://gazebosim.org/api/transport/11.0/installation.html) to install `libignition-transport12-dev`, and I also checked to have libgz-transport12-dev installed, the rosdep key does not resolve. Any tips?

retinfai, 2023-02-09:
The rosdep index hasn't been updated with the latest versions see [here](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml).  This isn't a ros_gz issue, so I recommend closing the issue – but relatively new to this so do as you please

Ryanf55, 2023-02-20:
> 

Sorry for the delay, I filed another issue with `gz-transport-12` and issued a fix for this. IMO, don't add rosdep keys and merge them if they don't exist yet, that's my issue with `ros_gz`

azeey, 2023-03-02:
We've added more Gazebo-ROS [documentation](https://gazebosim.org/docs/garden/ros_installation#gazebo-garden) including how to use the rosdep keys. 

> IMO, don't add rosdep keys in your package that don't exist and merge it. 

Isn't it better to have it in the package so you know exactly what the dependencies are. Otherwise, you'd have to go somewhere else to find that information. Anyway, since the rosdep keys can actually be used now, I think it's better to leave them in.

I'll close this issue since the problem was related to installing dependencies.

closed 2023-03-02 by azeey