gazebosim/ros_gz #360: Re-enable ros_gz_point_cloud

mjcarroll, 2023-02-07:
The point cloud plugin quite some time ago, but still has a lot of utility. The context for originally disabling is here: https://github.com/gazebosim/ros_gz/issues/40

mjcarroll, 2023-02-07:
CC: @bperseghetti

SuperJappie08, 2024-11-29:
I attempted to port this for ROS Jazzy and Gazebo Harmonic. (Because it looked useful and doable, and I wanted more about Gazebo)

However, even after converting it to a rendering plugin, the performance improvements gazebosim/gz-sensors#447 and gazebosim/gz-sensors#413, it is no longer possible to get data via the renderer directly without subscribing to the (gazebo) topic.

(My branch lives here, although not functioning as expected [SuperJappie08/ros_gz jazzy-point-cloud](https://github.com/SuperJappie08/ros_gz/tree/jazzy-point-cloud))