gazebosim/ros_gz #362: gz-sim plugin not loading

napat35253, 2023-02-09:
## Environment
* OS Version: MacOS M2, parallels VM Ubuntu 22.04
* Source or binary build? : source humble
  
  

## Description
* Expected behavior:  gz-sim plugin loaded
* Actual behavior: 
```
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-sensors-system] : couldn't find shared library.
```

## Steps to reproduce

1. use `ros_gz_sim` to launch the world
2. ros2 launch ros_gz_sim gz_sim.launch.py gz_args:="test.sdf"

## Output

```
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-sensors-system] : couldn't find shared library.
```
[image]

If I change the world file from for example 
```
      <plugin
        filename="gz-sim-sensors-system"
        name="gz::sim::systems::Sensors">
        <render_engine>ogre2</render_engine>
      </plugin>
```
to
```
      <plugin
        filename="ignition-gazebo-sensors-system"
        name="ignition::gazebo::systems::Sensors">
        <render_engine>ogre2</render_engine>
      </plugin>
```

it seems to work but I'm not sure should this is the way to solve this problem or not.

ahcorde, 2023-02-09:
@napat35253, which Gazebo version are you using ? Fortress? Garden ?

napat35253, 2023-02-09:
> @napat35253, which Gazebo version are you using ? Fortress? Garden ?

I'm using Garden

iche033, 2023-02-09:
From the error messages, you may have Fortress installed instead of Garden (or may be both?). We changed `ignition` and `ign` to `gz` in Garden. It seems like `ros_gz_sim` is loading Fortress in your case.

napat35253, 2023-02-09:
@iche033 I've tried uninstall and re-install following this https://gazebosim.org/docs/garden/install_ubuntu tutorial but it seems to have the same issue. Also, I try to setup a docker container for this and it has the same issue

Sebastian-Schroder, 2023-09-01:
I can concur but i have no extra information to add sadly.
I was required to change the plugin filename and name to the older ignition naming scheme despite having gazebo garden, and no file on my computer with that specific file name.
```xml
    <plugin
      filename="libgz-sim7-joint-state-publisher-system.so"
      name="gz::sim::systems::JointStatePublisher">
    </plugin>
```

-->

``` xml
    <plugin
      filename="ignition-gazebo-joint-state-publisher-system"
      name="ignition::gazebo::systems::JointStatePublisher">
    </plugin>
```

zhikh1, 2024-02-05:
Just specify the gz version explicitly in the ros_gz_sim launch args  
` launch_arguments=[`  ('gz_version', '7'),... `
Then the ros_gz_sim launch file will use gz instead of ign