gazebosim/ros_gz #363: SpawnEntity, DeleteEntity, SetPose Services

retinfai, 2023-02-10:
## Desired behavior
ros_gz_bridge support SpawnEntity, DeleteEntity, and SetPose services, so I can access them through ROS2 client calls.

## Alternatives considered
Running gz services through Subprocesses

## Implementation suggestion
Is it just `ros_gz_bridge/src/service_factories/ros_gz_interfaces.cpp` that needs to be changed? Not too sure, but just a thought.

## Additional context
Gazebo Garden, ROS2 Humble.
The interfaces for the services, request types, and response types are already supported in `ros_gz_interfaces` or other

azeey, 2023-02-16:
This would be a nice addition. A PR would be welcome 😉 .

retinfai, 2023-02-16:
I can give it a good go. First open source contribution attempt aha

zp-yang, 2023-03-01:
How is the progress on implementing these services? I would also like to help if needed.

retinfai, 2023-03-01:
I've begun work on this fork and hacked a solution for my own purposes here: https://github.com/retinfai/ros_gz

But more tweaking is needed before I want to submit for review.

Would love collaboration – I haven't found a lot of time recently.

zp-yang, 2023-03-04:
which ros2 version and gazebo version are you using? I am running humble and garden.

retinfai, 2023-03-04:
I'm using that too.

**ROS2 Humble** & **Gazebo Garden**

zp-yang, 2023-03-04:
Cool, what do you think still needs to be done? You seemed to have all of the features working. I am mainly interested in the set_pose service and it works on my side as well.

retinfai, 2023-03-04:
It's more like it's incomplete. I don't convert every field. 

But for set pose specifically, it gets confusing for me when gz.msgs.Pose doesn't map one to one to anything in the ROS side, so want to know the best practice for mapping 1 to N and vice versa. Set pose request (Entity and Pose on the ROS side) map to gazebo Pose but not all fields in Entity and Pose are used, so I exclude them. That sort of thing

azeey, 2023-03-08:
> I was wondering how to set the location of an entity spawned within Gazebo Garden. But it's not officially supported.

This is definitely supported. Checkout the message definition https://github.com/gazebosim/gz-msgs/blob/gz-msgs9/proto/gz/msgs/entity_factory.proto#L65

retinfai, 2023-03-09:
> @retinfai I'm not an expert but to call a gazebo service from a ros node, I need the bridge support ?
> 
> If not, I'm going to investigate inside the set_pose service, you mentionned !

That's correct, calling the service from a Ros node does require bridge support . Just wanted to clarify. 

In saying that, you can pull the fork, which does support it, while I fix issues and create a PR

Mildred34, 2023-03-10:
@retinfai Yes I'll do that! Thanks for the help

Trana, 2024-10-03:
I tested the create/remove/set_pose services on Jazzy and Harmonic and the changes in this PR worked fine for me.

j-rivero, 2025-03-13:
:information_source:  The Gazebo core team has included the review of PR into a community effort to attract new contributions, [see the message in LinkdIn](https://www.linkedin.com/posts/open-source-robotics-foundation_qos-docs-not-being-generated-in-sphinx-activity-7304943865529057281-7wFO?utm_source=share&utm_medium=member_desktop&rcm=ACoAABJBGkgBZvpxU2P6e4d-EBpI5IkiXVtoUVQ). I wrote some context and introduction about how to help in https://gist.github.com/j-rivero/9ea1abe2441c8dbf3cc4dffccfbdaff8.

elmeripk, 2025-07-23:
@PerFrivik Works correctly for me with jazzy+harmonic. Just be sure to update your packages.

PR #380 (2025-03-18): Add SpawnEntity, DeleteEntity, & SetEntityPose Support
# 🎉 New feature

Closes #363

## Summary
Add support for SpawnEntity, DeleteEntity, & SetEntityPose services
Add support for EntityFactory

## Test it
Spin up a parameter bridge for the service you are testing, and make a ros2 service call to it

Example for `SpawnEntity`
```
gz sim empty.sdf
ros2 run ros_gz_bridge parameter_bridge /world/empty/create@ros_gz_interfaces/srv/SpawnEntity
ros2 service call /world/empty/create ros_gz_interfaces/srv/SpawnEntity "{entity_factory: {name: "cool_sdf", sdf_filename: 'path/to/sdf'}}"
```

## Checklist [... cut]
PR #712 (2025-03-20): Add SpawnEntity, DeleteEntity, & SetEntityPose Support (backport #380)
# 🎉 New feature

Closes #363

## Summary
Add support for SpawnEntity, DeleteEntity, & SetEntityPose services
Add support for EntityFactory

## Test it
Spin up a parameter bridge for the service you are testing, and make a ros2 service call to it

Example for `SpawnEntity`
```
gz sim empty.sdf
ros2 run ros_gz_bridge parameter_bridge /world/empty/create@ros_gz_interfaces/srv/SpawnEntity
ros2 service call /world/empty/create ros_gz_interfaces/srv/SpawnEntity "{entity_factory: {name: "cool_sdf", sdf_filename: 'path/to/sdf'}}"
` [... cut]
PR #713 (2025-03-20): Add SpawnEntity, DeleteEntity, & SetEntityPose Support (backport #380)
# 🎉 New feature

Closes #363

## Summary
Add support for SpawnEntity, DeleteEntity, & SetEntityPose services
Add support for EntityFactory

## Test it
Spin up a parameter bridge for the service you are testing, and make a ros2 service call to it

Example for `SpawnEntity`
```
gz sim empty.sdf
ros2 run ros_gz_bridge parameter_bridge /world/empty/create@ros_gz_interfaces/srv/SpawnEntity
ros2 service call /world/empty/create ros_gz_interfaces/srv/SpawnEntity "{entity_factory: {name: "cool_sdf", sdf_filename: 'path/to/sdf'}}"
` [... cut]