gazebosim/ros_gz #365: Parameter_bridge does not work

rows-of-houses, 2023-02-21:
It seems that parameter_bridge does not work in my case. I have:
* Ubuntu 22.04
* ROS2 Humble
* Gazebo Garden
* caguero/humble_garden branch (I cannot use ros2 branch instead because it crushes while being built)

I want to send data from IMU to ROS2 topic. 

I have three terminals:
1. I run my gz world with `gz sim world.sdf` and run simulation
2. I make `cd` to my bridge workspace and run `. install/setup.bash`, then run `ros2 run ros_gz_bridge parameter_bridge /imu/data_raw@sensor_msgs/msg/Imu[gz.msgs.IMU` (/imu/data_raw is actual topic of IMU data transition, I can see this with `gz topic -e -t /imu/data_raw`
3. I make `cd` to my bridge workspace and run `. install/setup.bash`. There is a topic /imu/data_raw in ROS2 (I checked it with `ros2 topic list`), but if I try to read data from it with `ros2 topic echo /imu/data_raw`, there is no any messages.

YDenker, 2023-02-22:
I have an issue with one of my parameter bridges as well that worked just fine before the renaming. All others work just fine.
The bridge that is not working for me is the following:
ros2 run ros_gz_bridge parameter_bridge /collision_detection@ros_ign_interfaces/msg/Contacts[ignition.msgs.Contacts
I am using

Ubuntu 22.04.2 LTS
ROS2 Humble 
Gazebo Citadel

I get the warning that the bridge is not being created:
[WARN] [1677058258.983101520] [ros_gz_bridge]: Failed to create a bridge for topic [/collision_detection] with ROS2 type [/collision_detection] to topic [ros_ign_interfaces/msg/Contacts] with Gazebo Transport type [ignition.msgs.Contacts]

Did something change with the transport types that i am unaware of?

rows-of-houses, 2023-02-24:
I tried using `humble` branch, but it also didn't work.
After 
`ros2 run ros_gz_bridge parameter_bridge /imu/data_raw@sensor_msgs/msg/Imu[gz.msgs.IMU`
 I got
`[INFO] [1677239017.350256340] [ros_gz_bridge]: Creating GZ->ROS Bridge: [/imu/data_raw (gz.msgs.IMU) -> /imu/data_raw (sensor_msgs/msg/Imu)] (Lazy 0)`, 
but there is still no messages in `/imu/data_raw` ROS2 topic. But I see messages in Gazebo topic.

azeey, 2023-02-28:
I would suggest using the `humble` branch since the `caguery/humble_garden` branch is quite outdated now (https://github.com/gazebosim/ros_gz/compare/caguero/humble_garden..humble). If your build crashes, it's probably due to insufficient memory and following the suggestion in https://github.com/gazebosim/ros_gz/issues/235#issuecomment-1093124301 might help

doppioandante, 2023-03-07:
I am having the same problem, using the `humble` branch, the bridge seems to actually make the topics visible from either side but no message is being delivered, I'm trying the standard chatter message.
Btw the doc is outdated and uses ignition message types

doppioandante, 2023-03-08:
This is a reproducible Dockerfile with the configuration and build steps I'm using:
```Dockerfile
FROM osrf/ros:humble-desktop

RUN apt-get update
RUN apt-get install kakoune wget gnupg apt-utils -y
RUN wget https://packages.osrfoundation.org/gazebo.gpg -O /usr/share/keyrings/pkgs-osrf-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-stable.list > /dev/null
RUN apt-get update

RUN apt-get install gz-sim7-cli libgz-transport12-dev libgz-math7-dev libgz-cmake3-dev libignition-cmake2-dev -y

RUN useradd --create-home --shell /bin/bash ros2

USER ros2
WORKDIR /home/ros2
RUN rosdep update
RUN mkdir -p /home/ros2/ws/src/
RUN cd /home/ros2/ws/src/ && git clone --depth 1 -b humble https://github.com/gazebosim/ros_gz.git 

USER root
WORKDIR /home/ros2/ws
RUN rosdep install -r --from-paths src -i -y --rosdistro humble

RUN echo "source /opt/ros/humble/setup.bash" >> /home/ros2/.bashrc
WORKDIR /home/ros2/ws
RUN bash -c "source /opt/ros/humble/setup.bash && colcon build"

RUN apt-get install gz-garden -y
USER ros2
RUN echo "source /home/ros2/ws/install/setup.bash" >> /home/ros2/.bashrc
```

azeey, 2023-03-08:
Can you try exporting the Gazebo version (`GZ_VERSION`) per https://github.com/gazebosim/ros_gz/tree/humble#gazebo

azeey, 2023-03-08:
> Wow thanks, I had missed that. It seems to be working now, at least the basic chatter message.

Glad it works! And thanks for the reproducible Dockerfile.

> As a side note, rosdep doesn't seem so good at finding all the dependencies, I had to install some dev packages manually, but I'm very new to the whole ROS ecosystem so there may be something missing.

We need to link this in the readme here, but there are some instructions for `rosdep` issues at https://gazebosim.org/docs/garden/ros_installation

@rows-of-houses can you try my suggestion in https://github.com/gazebosim/ros_gz/issues/365#issuecomment-1460937619

tommysmth, 2023-04-19:
I am having the same problem, using also the humble branch. I get a message output by gazebo but none with the ROS topic which is created. So the bridge is creating the topic without passing messages on to it. I've had the problem for geometry_msgs/msg/PoseArray | gz.msgs.Pose_V and sensor_msgs/NavSatFix | gz.msgs.NavSat where the topics are created, but no messages reach through. It's getting even worse when I'm using tf2_msgs/TFMessage | gz.msgs.Pose_V and gps_msgs/GPSFix | gz.msgs.NavSat instead. For those, I'm not even getting a topic on ROS.

azeey, 2023-05-01:
@tommysmth Have you built `ros_gz` from source using `GZ_VERSION=garden`?

azeey, 2023-05-26:
Closing since the issue was resolved by setting `GZ_VERSION=garden` when compiling `ros_gz`.

closed 2023-05-26 by azeey